eclipse-mraa/mraa #88: Illegal arguments for construction of _exports_Gpio Edison on mini breakout with NodeJS

jamidwyer, 2015-01-07:
Basic blink code gives me this error:
MRAA Version: v0.5.4-64-g3efb7a0

/home/root/b.js:13
var myLed = new m.Gpio(12);
            ^
Error: Illegal arguments for construction of _exports_Gpio
    at Object.<anonymous> (/home/root/b.js:13:13)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:906:3

I see an open issue for the same error under different circumstances, so feel free to close this if they are related. Thank yoU!

arfoll, 2015-01-07:
The other issue is very different. Gpio12 isn't a valid pin check our pin table - https://github.com/intel-iot-devkit/mraa/blob/master/docs/edison.md. Not sure what you meant maybe J17-12 which would be pin11? Please check the output on syslog using `journalctl -f` whilst running your code.

Please close if fixed.

jamidwyer, 2015-01-07:
Well how about that. I must be reading the GPIO to pin conversion chart wrong. Thank you for such quick help!

closed 2015-01-07 by jamidwyer

jamidwyer, 2015-01-07:
Nothing wrong with the chart at all -- don't know where I got the idea that J18 pin 2 was GPIO 12. Thanks again.

arfoll, 2015-01-07:
J is the 10th letter in the alphabet + 2? :)

Have fun using your Edison board! Please excuse our hardware guys who
decided labelling pads with numbers was too obvious :/

We do have an enum I recently added to try make it easier:
https://github.com/intel-iot-devkit/mraa/blob/master/api/mraa/types.h#L64
So it should work from node wihtout the MRAA_ prefix.