eclipse-mraa/mraa #91: Node.js PWM should disable pin when 0 is written

Fleker, 2015-01-11:
If one writes 0 to a PWM pin, it should automatically disable that pin.

    var pin = new mraa.Pwm(pin, -1, false);
    pin.enable(false) // Should be called if PWM is 0. 

I'm working with some LEDs and they keep showing through even when they have a value of 0. They only stop working when manually disabled.

Also, the Node.js API is poorly documented. I'm not too sure what each part of the constructor is. I'm assuming they are the pin number, `period_us`, and enabled.

arfoll, 2015-01-11:
Hi @Fleker, try the C++ API docs - http://iotdk.intel.com/docs/master/mraa/classmraa_1_1_pwm.html#a234a0b528c55d2157de74abf4b38daa8 for the argument description.

You should really only be providing the first constructor parameter unless you understand what owner/chipid means. Note those two args will likely get switched around since 'owner' is a prop people will want to set on a non raw pwm object. See #93.

The API docs, I'm working on we'll publish them again soon. I've opened #92.

Let's keep this bug for discussion on just what you first mention - we should disable the pin when writing 0.

miminashi, 2016-01-19:
I also facing same problem with IntelEdison + mraa 0.9.0 Python binding.
I'm not sure if it is hardware's matter or mraa's matter.
If it is mraa matter, I wish this issue will be solved.

`write(0)` to 3 LEDs, rightmost LED is `enabled(False)`.
[image]

alext-mkrs, 2016-05-17:
I've started looking into this yesterday and looks like there's a bigger problem with either the pwm IC driver or the IC itself. Every couple of enable/set duty/disable ends with a pin at what seems to be a 100% duty, instead of disabled. At that stage any further enable/disable attempts won't work and only setting other duty cycle gets it out of this state.

I have to look further into this, but if that's an Edison-specific thing, and the one that seems to be broken in several places, maybe we better fix it or do it in a board-specific way rather than changing semantics of duty cycle for everyone.

@arfoll, what do you think?

alext-mkrs, 2016-05-19:
I've checked the driver with some debug output enabled and this might even be a HW bug/limitation, as the driver reports that it honestly sets the enable/disable bit, but every now and then (but not 100% repeatable) that ends in a 100% duty instead of disabled.

In view of that adding a forced pin disable upon `write(0.0)` makes no sense.

@arfoll, do you have any means to report/check it out with someone for the case if that's a known limitation/problem? We could probably contact the PWM device driver author from Emutex as well.

alext-mkrs, 2016-05-24:
So finally - the 0% duty thing looks like a known  HW problem, so I've proposed a workaround, which doesn't alter the global API/flow, which is IMHO unnecessary here.

The "100%-instead-of-disable" problem looks like a very different thing and also HW-related. I'll open a separate issue for this and will investigate it further, but I don't think it by any means blocks resolving this one.

commit 8af68435 (2016-06-16): PWM: added workaround for Edison's problem with 0% duty

As described in issue #91, on Edison setting 0% duty doesn't
disable the PWM on a pin completely.

Therefore we add a couple of Edison-specific _pre functions
and an internal PWM state variable, which we use to toggle PWM
enabled/disabled based on what duty is set for the pin.

Closes #91.
closed 2016-06-16 by arfoll