eclipse-mraa/mraa #99: galileo gen1 initalisation is treated as 'error'

arfoll, 2015-01-19:
There is no 'correct' gen1 initialisation anymore meaning we print an error message even if it isn't the case.

https://github.com/intel-iot-devkit/mraa/blob/master/src/x86/x86.c#L63

alext-mkrs, 2015-01-25:
Just to make sure I understand the intent - we simply don't want to log this a namely error (but e.g. notice) or we want to change the init logic and e.g. stop using Gen1 clause as a fallback?

Feel free to assign this one to me, BTW - I'll drive it to closure.

arfoll, 2015-01-25:
@alext-mkrs gen1 as fallback is ok I think, I dont have a better idea for now so I don't want to change the behaviour for now.

The issue is that there there should be an `else if (strncasecmp(line, "Galileo", x)` and then the default should do the same but also log an error. Does that make sense?

alext-mkrs, 2015-01-25:
Yep, it does, thanks. Let me implement that.

commit e68546ad (2015-01-26): x86.c/mraa_x86_platform: added dedicated block for Galileo Gen1 init

Closes #99.
closed 2015-01-26 by tingleby