eclipse-mraa/mraa #102: d.ts header for libmraa

arfoll, 2015-01-20:
As discussed in #101 by @barais it would be great to provide a typeface d.ts header to libmraa.

barais, 2015-01-21:
Ok I just release here the d.ts generator from hpp header

https://github.com/HEADS-project/mraa_hpp2ts_generator

I you disagree with the mapping of type you just have to change that file.

https://github.com/HEADS-project/mraa_hpp2ts_generator/blob/master/src/main/java/eu/heads/project/ts/HPPSourceParser.java#L61

arfoll, 2015-01-25:
That is quite cool, however I feel it's too complicated to add to our build system. How about I add this to our CI system so every time we make a new build it generates a d.ts file and uploads it somewhere so people can grab it?

The mapping of types still seems a bit off, I'm not too sure how your program works but uint8_t pointers should get mapped to a node_buffer which I'm not sure is supported in general js? But that's what we expect as an input in mraa.node.