eclipse-mraa/mraa #105: Don't check for root at initialization

eschaton, 2015-01-25:
Right now it looks like `libmraa` checks whether it's in a process with effective uid 0 at initialization. In order to implement least-privilege access control, I'd like to restrict access to I/O managed by `libmraa` to just members of a specific group.

It really shouldn't be `libmraa`'s job to perform this check, especially pre-emptively. Since this check was probably added to help troubleshooting, the same goal could be achieved by logging the permissions when `libmraa` fails to open a device file due to `EPERM`.

alext-mkrs, 2015-01-25:
@arfoll, @tingleby I'd be curious to have your opinion here gentlemen, but my personal understanding is that it's currently not feasible to run mraa as non-EUID 0, because that would mean a lot of messing up with [at least] permissions on GPIO files for the user. That's why we've added it in the first place.

tingleby, 2015-01-25:
I was under the impression that it it wouldn't be possible to setup permissions under the sysfs to use gpio for non root users to write into it.
There would be more luck in the modules that use /dev/ (i2c/spi) to run as non root with permissions set.

alext-mkrs, 2015-01-25:
Right, that's what I had in mind as well. So @eschaton it looks like it's simply not possible to do that.

If you have any specific scenario in mind, please share and let's see if something could be done here and generalized for the benefit of more use cases.

arfoll, 2015-01-25:
@eschaton I agree it's not really our job, we do it because it makes our debugging life much easier :) I guess it should just log it as a warning and see how far we can go. But I'd be really interested in how you can setup your system to handle not running as root, could you detail that so we can maybe document it for our users?

@tingleby that's not entirely true, AFAIK you can chown/chmod sysfs properties. The problem is that created nodes will still be under the default perms root:root. This means you'd need a rather complex and dodgy udev rule to run on every gpio export (for example). When it comes to writing to /dev/mem or /dev/uio as we do for mmaped gpio giving non root access to those is basically the same as giving the keys away IMHO and there is a kernel capability for that. As you say i2c/spi is all good (as long as we have no muxing), we only really have this issue with gpio & pwm.

tingleby, 2015-01-25:
@arfoll Yes making it a bit dodgy. What I would be afraid of would be users with permissions twiddling a gpio that shouldn't be touched. Ones that could reset/damage a system.

eschaton, 2015-01-31:
Thanks everyone for the feedback!

@arfoll I haven't actually set it up that way, I was going to try setting it up just by doing a `chown` of the sysfs files to a `gpio` group and running the process as a user which is a member of that group.

@alext-mkrs Mainly I want privilege separation, and I figured that without capabilities that user/group would be the next logical mechanism. I really don't want a process interfacing with the real world and possibly the network running as root if I can help it.

I don't know much about things like Linux `sysfs` so I'm not sure about the gymnastics that would be needed to actually make user/group access control work with it. However, I'd really only expect file-based access to work with user/group access control, and that to be sufficient for a lot of uses.

For memory mapped I/O, I'd expect the driver to support some sort of query where it can hand back to the user process a limited chunk of memory it (and only it) can manipulate, and for doing so to involve an access control check. I wouldn't expect to have access to the entire memory map. (If you can't tell yet, I've mostly worked with what you might call "Darwin-based" systems…)

arfoll, 2015-01-31:
@eschaton You convinced me, it's not our job to do this check. It's removed with baa1a0ae03aca521d2fb5a3e043fe5ef02a52b36

As for mmaped IO sadly alot of the drivers we find don't do things in 'sane' ways so setting up non root access to the IO may be more difficult on some platforms than others. Personally I'd recommend abstracting your IO at a high-ish level with a daemon running itself as root with something like dbus to talk to it from a non priviliged process, that should give you a decent amount of flexibility without having to run everything as root.

closed 2015-01-31 by arfoll