eclipse-mraa/mraa #108: running mraa(for js) repeatedly will cause the segmentation fault or strange error

ddnionio, 2015-01-27:
mraa: 0.5.4
platform: edison ww42
nodejs: 0.10.28

code:
```javascript
var io = require("mraa");
var interval = 1000;

setInterval(digitalWrite,interval ,0);
setInterval(digitalWrite,interval ,1);
setInterval(digitalWrite,interval ,2);
setInterval(digitalWrite,interval ,3);
setInterval(digitalWrite,interval ,4);
setInterval(digitalWrite,interval ,5);
setInterval(digitalWrite,interval ,6);

function digitalWrite(pin){
  var p = new io.Gpio(pin);
  p.dir(io.DIR_OUT);
  p.write(1);
  console.log(pin);
}
```
(you can change the interval form 1000 to 10, in order to see the error soon)
After running some time, it shows __"segmantation fault" or__
```
undefined:0
TypeError: undefined is not a function
```
**or traceinfo as follows**
```
*** Error in `node': munmap_chunk(): invalid pointer: 0x09966080 ***
======= Backtrace: =========
/lib/libc.so.6[0x4c460f2d]
/lib/libc.so.6[0x4c46bad9]
/lib/libc.so.6[0x4c46bbac]
/usr/lib/libstdc++.so.6(_ZdlPv+0x21)[0x4c87f271]
/usr/lib/node_modules/mraa/mraa.node(_Z17_wrap_delete_GpioN2v810PersistentINS_5ValueEEEPv+0x63)[0xb7732cdf]
node(_ZN2v88internal13GlobalHandles31PostGarbageCollectionProcessingENS0_16GarbageCollectorE+0xe0)[0x81cbe40]
node(_ZN2v88internal4Heap24PerformGarbageCollectionENS0_16GarbageCollectorEPNS0_8GCTracerE+0x1ed)[0x81e15bd]
node(_ZN2v88internal4Heap14CollectGarbageENS0_15AllocationSpaceENS0_16GarbageCollectorEPKcS5_+0x138)[0x81e2dc8]
node(_ZN2v88internal7Factory37BaseNewFunctionFromSharedFunctionInfoENS0_6HandleINS0_18SharedFunctionInfoEEENS2_INS0_3MapEEENS0_13PretenureFlagE+0xce)[0x81b64ee]
node(_ZN2v88internal7Factory33NewFunctionFromSharedFunctionInfoENS0_6HandleINS0_18SharedFunctionInfoEEENS2_INS0_7ContextEEENS0_13PretenureFlagE+0x77)[0x81b6727]
node(_ZN2v88internal18Runtime_NewClosureENS0_9ArgumentsEPNS0_7IsolateE+0x12e)[0x82e6dbe]
[0x2c20a236]
[0x35d21262]
[0x2c20f7a1]
[0x35d1c419]
[0x2c20f7a1]
[0x35d1bd68]
[0x2c20f7a1]
[0x2c20fe12]
[0x2c20a0aa]
node[0x81a8723]
node(_ZN2v88internal9Execution4CallENS0_6HandleINS0_6ObjectEEES4_iPS4_Pbb+0xc6)[0x81a9dc6]
node(_ZN2v88internal13Runtime_ApplyENS0_9ArgumentsEPNS0_7IsolateE+0x195)[0x82ebea5]
[0x2c20a236]
[0x2c2689a5]
[0x2c20f7a1]
[0x35d18dac]
[0x2c2141a5]
[0x35d182ea]
[0x2c25b274]
[0x2c20f7a1]
[0x2c20fe12]
[0x2c20a0aa]
node[0x81a8723]
node(_ZN2v88internal9Execution4CallENS0_6HandleINS0_6ObjectEEES4_iPS4_Pbb+0xc6)[0x81a9dc6]
node(_ZN2v88Function4CallENS_6HandleINS_6ObjectEEEiPNS1_INS_5ValueEEE+0x11d)[0x815717d]
node(_ZN4node12MakeCallbackEN2v86HandleINS0_6ObjectEEENS1_INS0_8FunctionEEEiPNS1_INS0_5ValueEEE+0x59)[0x83d8e99]
node(_ZN4node12MakeCallbackEN2v86HandleINS0_6ObjectEEENS1_INS0_6StringEEEiPNS1_INS0_5ValueEEE+0x5a)[0x83d907a]
node[0x8400630]
node(uv__run_timers+0x6e)[0x8533fde]
node(uv_run+0x85)[0x8529b95]
node(_ZN4node5StartEiPPc+0x18d)[0x83db20d]
node(main+0x1b)[0x81441bb]
/lib/libc.so.6(__libc_start_main+0xe7)[0x4c410687]
node[0x8144434]
======= Memory map: ========
08048000-087ab000 r-xp 00000000 b3:08 2169       /usr/bin/node
087ab000-087b7000 rwxp 00763000 b3:08 2169       /usr/bin/node
087b7000-087bd000 rwxp 00000000 00:00 0
087bd000-08864000 rwxp 0076e000 b3:08 2169       /usr/bin/node
09938000-09a73000 rwxp 00000000 00:00 0          [heap]
24d00000-24d39000 rwxp 00000000 00:00 0
29a03000-29a04000 r-xp 00000000 00:00 0
2ae99000-2ae9a000 r-xp 00000000 00:00 0
2c200000-2c209000 rwxp 00000000 00:00 0
2c209000-2c20a000 ---p 00000000 00:00 0
2c20a000-2c26a000 rwxp 00000000 00:00 0
2c26a000-2c26b000 ---p 00000000 00:00 0
33c00000-33d00000 rwxp 00000000 00:00 0
35d00000-35d09000 rwxp 00000000 00:00 0
35d09000-35d0a000 ---p 00000000 00:00 0
35d0a000-35dff000 rwxp 00000000 00:00 0
35dff000-35e00000 ---p 00000000 00:00 0
39000000-39049000 rwxp 00000000 00:00 0
3aa25000-3aa26000 r-xp 00000000 00:00 0
3aded000-3adee000 r-xp 00000000 00:00 0
3f100000-3f200000 rwxp 00000000 00:00 0
4be00000-4be19000 rwxp 00000000 00:00 0
4c3c8000-4c3e8000 r-xp 00000000 b3:08 1575       /lib/ld-2.19.so
4c3e8000-4c3e9000 r-xp 0001f000 b3:08 1575       /lib/ld-2.19.so
4c3e9000-4c3ea000 rwxp 00020000 b3:08 1575       /lib/ld-2.19.so
4c3ec000-4c3ef000 r-xp 00000000 b3:08 2015       /lib/libdl-2.19.so
4c3ef000-4c3f0000 r-xp 00002000 b3:08 2015       /lib/libdl-2.19.so
4c3f0000-4c3f1000 rwxp 00003000 b3:08 2015       /lib/libdl-2.19.so
4c3f7000-4c566000 r-xp 00000000 b3:08 2025       /lib/libc-2.19.so
4c566000-4c567000 ---p 0016f000 b3:08 2025       /lib/libc-2.19.so
4c567000-4c569000 r-xp 0016f000 b3:08 2025       /lib/libc-2.19.so
4c569000-4c56a000 rwxp 00171000 b3:08 2025       /lib/libc-2.19.so
4c56a000-4c56d000 rwxp 00000000 00:00 0
4c56f000-4c587000 r-xp 00000000 b3:08 1629       /lib/libpthread-2.19.so
4c587000-4c588000 r-xp 00017000 b3:08 1629       /lib/libpthread-2.19.so
4c588000-4c589000 rwxp 00018000 b3:08 1629       /lib/libpthread-2.19.so
4c589000-4c58b000 rwxp 00000000 00:00 0
4c5a5000-4c5f0000 r-xp 00000000 b3:08 1648       /lib/libm-2.19.so
4c5f0000-4c5f1000 r-xp 0004a000 b3:08 1648       /lib/libm-2.19.so
4c5f1000-4c5f2000 rwxp 0004b000 b3:08 1648       /lib/libm-2.19.so
4c5f4000-4c5fb000 r-xp 00000000 b3:08 2013       /lib/librt-2.19.so
4c5fb000-4c5fc000 r-xp 00006000 b3:08 2013       /lib/librt-2.19.so
4c5fc000-4c5fd000 rwxp 00007000 b3:08 2013       /lib/librt-2.19.so
4c81e000-4cAborted
```

arfoll, 2015-01-27:
I have an idea but need to check with a debugger. Could you post the output of `journalctl -t libmraa` or use `journalctl -f` whilst running your code.

Honestly you should never use mraa like that because performance will be poor. You should always reuse the gpio context and avoid having multiple ones for the same pins.

rwaldron, 2015-01-27:
> Honestly you should never use mraa like that because performance will be poor. You should always reuse the gpio context and avoid having multiple ones for the same pins.

In my experience, up-front allocation of all the Gpio objects that the program will use gives the best perceptible performance. If I have a chance I will do some benchmarking and memory profiling.

ddnionio, 2015-01-28:
@arfoll 
sorry that I don't know well about "journalctl", it shows "invalid option -- 't'", and -f shows: (i think it is not what you need)
```

root@hope3:~# journalctl -f
-- Logs begin at Sat 2000-01-01 00:00:10 UTC. --
Jan 28 05:12:34 hope3 systemd-timesyncd[158]: Using NTP server 216.239.32.15:123 (time1.google.com).
Jan 28 05:12:45 hope3 systemd-timesyncd[158]: Timed out waiting for reply from 216.239.32.15:123 (time1.google.com).
Jan 28 05:12:45 hope3 systemd-timesyncd[158]: Using NTP server [2001:4860:4802:32::f]:123 (time1.google.com).
Jan 28 05:12:45 hope3 systemd-timesyncd[158]: Using NTP server 216.239.34.15:123 (time2.google.com).
Jan 28 05:12:55 hope3 systemd-timesyncd[158]: Timed out waiting for reply from 216.239.34.15:123 (time2.google.com).
Jan 28 05:12:55 hope3 systemd-timesyncd[158]: Using NTP server [2001:4860:4802:34::f]:123 (time2.google.com).
Jan 28 05:12:55 hope3 systemd-timesyncd[158]: Using NTP server 216.239.36.15:123 (time3.google.com).
Jan 28 05:13:05 hope3 systemd-timesyncd[158]: Timed out waiting for reply from 216.239.36.15:123 (time3.google.com).
Jan 28 05:13:05 hope3 systemd-timesyncd[158]: Using NTP server [2001:4860:4802:36::f]:123 (time3.google.com).
Jan 28 05:13:05 hope3 systemd-timesyncd[158]: Using NTP server 216.239.38.15:123 (time4.google.com).
Jan 28 05:13:15 hope3 systemd-timesyncd[158]: Timed out waiting for reply from 216.239.38.15:123 (time4.google.com).
Jan 28 05:13:15 hope3 systemd-timesyncd[158]: Using NTP server [2001:4860:4802:38::f]:123 (time4.google.com).
Jan 28 05:13:15 hope3 systemd-timesyncd[158]: Using NTP server 216.239.32.15:123 (time1.google.com).
Jan 28 05:13:26 hope3 systemd-timesyncd[158]: Timed out waiting for reply from 216.239.32.15:123 (time1.google.com).
Jan 28 05:13:26 hope3 systemd-timesyncd[158]: Using NTP server [2001:4860:4802:32::f]:123 (time1.google.com).
Jan 28 05:13:26 hope3 systemd-timesyncd[158]: Using NTP server 216.239.34.15:123 (time2.google.com).
```

I know it is an abnormal usage to instantiate mraa object repeatedly, but in my case, it is needed.

arfoll, 2015-01-28:
Unfortunately, mraa is not thread safe and causing objects to be instantiated like that can lead to the garbage collector cleaning up resources whilst things are being created. Can you explain why you can't instantiate the objects first? I'm interested in the use case.

I'm still looking into the issue to see if we can't lock down creation on close but this will take a while.

arfoll, 2015-01-28:
Actually I think I know what happens, in your case to start with you'll need to set Owner to False, so use `Gpio(num, False)`. This will force mraa to not unexport a gpio fully even when the garbage collector starts. Typically in C/C++ when the destructor is called in a predicatable way this is not an issue but in js/python the gc will cause an unexport even if something else know expects to see the value (note the first gpio object can have been out of scope for ages, the v8 gc is notoriously lazy).

Can you confirm that setting owner to False fixes your issue?

ddnionio, 2015-01-28:
I will try it tommorow (because I am in China, and already off work).

We are developing a task manager. Users may write the task by mraa, such as turn on the light for 1s
```javascript
var m = require("mraa")
var p = new m.Gpio(1)// assume the led is in pin 1
p.dir(m.DIR_OUT);
p.write(1);
setTimeout(function(){p.write(0)},1000);
```
and then he chose to set the task will run repeatedly in every 10s. so the problem happened.

We don't plan to constrain users' coding, they should code as without our manager.

I hope i explained clearly, if not, just tell me.
Thanks

ddnionio, 2015-01-29:
@arfoll it still reports after running about 5s
```
undefined:0
TypeError: undefined is not function
```

my code now is as follows: (using __new io.Gpio(pin,false)__)
``` javascript
var io = require("mraa");
var interval = 10;

setInterval(digitalWrite,interval ,0);
setInterval(digitalWrite,interval ,1);
setInterval(digitalWrite,interval ,2);
setInterval(digitalWrite,interval ,3);
setInterval(digitalWrite,interval ,4);
setInterval(digitalWrite,interval ,5);
setInterval(digitalWrite,interval ,6);

function digitalWrite(pin){
var p = new io.Gpio(pin,false);
console.log(pin);
}
```

rwaldron, 2015-01-29:
> We are developing a task manager. Users may write the task by mraa, such as turn on the light for 1s
...
> We don't plan to constrain users' coding, they should code as without our manager.

Neither of these points precludes doing the allocation up front, which _will_ solve the problem that you're reporting. 

```js
require("es6-shim");

var io = require("mraa");
var pins = Array.from({ length: 7 }, function(_, index) {
  return new io.Gpio(index, false);
});
var interval = 10;

setInterval(digitalWrite,interval ,0);
setInterval(digitalWrite,interval ,1);
setInterval(digitalWrite,interval ,2);
setInterval(digitalWrite,interval ,3);
setInterval(digitalWrite,interval ,4);
setInterval(digitalWrite,interval ,5);
setInterval(digitalWrite,interval ,6);

function digitalWrite(pin) {
  console.log(pins[pin]);
  // ...
}
```

ddnionio, 2015-01-30:
@rwaldron 
you may misunderstand what i mean.

*The user of our tool will firstly write the implementation of the single action, like turning on the LED for 1s. They should write the single action just as they do without out tool. so they may use mraa and write 
```javascript
var m = require("mraa")
var p = new m.Gpio(1)// assume the led is in pin 1
p.dir(m.DIR_OUT);
p.write(1);
setTimeout(function(){p.write(0)},1000);
``` 

*If they want to implement a task that the single action will run repeatedly in 10s intervals, and they use our tool by providing the single action's code and setting relative config files.

*Our task manager tool will then "call" the single action repeatedly, which will cause the problem.

amersaw, 2015-02-09:
I had exactly the same problem when I used "oled-js" with "johny-five" with "edison-io" plugin, the "edison-io" module uses "mraa" and that error occurs when I call some output process using setTimeout().

I've tried to modify the "galileo.js" file that calls  

```javascript
    new Pin.IO.Gpio(1);
```
by default, and set the owner to false as @arfoll mentioned but also the same, I've tried later to set the owner to undefined variable `smmmmthing` for example 

```javascript
    new Pin.IO.Gpio(1,smmmmthing); //where smmmmthing is not defined in other place
```
 and it worked fine!!

rwaldron, 2015-02-09:
@amersaw can you file this with https://github.com/rwaldron/galileo-io? Thanks.

rwaldron, 2015-02-09:
@amersaw does `new Pin.IO.Gpio(1, undefined);` have the same result?

amersaw, 2015-02-09:
@rwaldron  Yes it works too.

ddnionio, 2015-02-10:
sorry, I am confused.  Does it solve my problem?

if I type 
```javascript
var p = new io.Gpio(pin,undefined);
```
it shows:
```
Error: Illegal arguments for construction of _exports_Gpio
    at digitalWrite (/home/root/test.js:16:9)
    at wrapper [as _onTimeout] (timers.js:258:14)
    at Timer.listOnTimeout [as ontimeout] (timers.js:110:15)
```

if i type
```javascript
var p = new io.Gpio(pin,haha);//haha is not defined before
```
it shows
```
ReferenceError: haha is not defined
    at digitalWrite (/home/root/test.js:16:25)
    at wrapper [as _onTimeout] (timers.js:258:14)
    at Timer.listOnTimeout [as ontimeout] (timers.js:110:15)
```

alext-mkrs, 2015-10-26:
@ddnionio it's been a while and both mraa and node.js were noticeably updated since your last post.

Do you still observe this problem and need help?

alext-mkrs, 2015-11-17:
Closing as no activity. Feel free to add a comment if you need further help.

closed 2015-11-17 by alext-mkrs

jeffdare, 2016-08-24:
Hi all,

I still see this issue with the latest mraa and node. 

This issue, does not always happen. I see that sometimes it works fine for hours, and sometimes get this issue as fast as 2-5 seconds. Is there any permanent fix for this?

arfoll, 2016-08-24:
Sorry this is basically ancient stuff now. Can you file a new issue with a code sample, what platform are you on the output of the logs as it crashes?

The issue here was (more than likely) due to node.js calling cleanup on a gpio which was later being re-used, essentially you should set owner to false if you open and close gpios many times and avoid doing it altogether.