eclipse-mraa/mraa #110: can't use ISR with javascript

oloynet, 2015-01-28:
I've try to adapt unsuccessfully the python example [hello_isr.py](https://github.com/intel-iot-devkit/mraa/blob/master/examples/python/hello_isr.py) to do the same with javascript with no chance.

```javascript
"use strict";

var m = require('mraa');
console.log('MRAA Version: ' + m.getVersion());

var counter   = 0;
var interrupt = 0;
var led_state = 0;
var argument  = 'myargument';

/*
function callback(arg) {
  interrupt++;
  console.log(arg);
}
*/

var callback = function(arg) {
  interrupt++;
  console.log(arg);
}

var myled    = new m.Gpio(3);
myled.dir(m.DIR_OUT);

var mybutton = new m.Gpio(2);
mybutton.dir(m.DIR_IN);
mybutton.isr(m.EDGE_BOTH, callback, argument);

function mainloop() {
  myled.write(led_state = led_state ? 0 : 1);
  counter++;
  console.log('counter = ' + counter + ' -- interrupt done = ' + interrupt);
  setTimeout(mainloop, 1000);
}

mainloop();
```
and I've the following error message
```bash
MRAA Version: v0.5.4-100-g6c73a8a

/home/root/examples/js/hello_isr.js:28
mybutton.isr(m.EDGE_BOTH, callback, argument);
         ^
Error: in method 'Gpio_isr', argument 3 of type 'void (*)(void *)'
    at Object.<anonymous> (/home/root/examples/js/hello_isr.js:28:10)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:906:3
```
If I change the following parameter by
```javascript
mybutton.isr(m.EDGE_BOTH, null, null);
```
The program start, and when I push the button, I've got a segmentation fault.
It's normal because the callback parameter is null.
```bash
MRAA Version: v0.5.4-100-g6c73a8a
counter = 1 -- interrupt done = 0
counter = 2 -- interrupt done = 0
counter = 3 -- interrupt done = 0
counter = 4 -- interrupt done = 0
Segmentation fault
```
After looking to ISR definition in library in C files, is there is a problem of definition the third argument or a problem with swig  ?
```c
mraa_gpio_isr(mraa_gpio_context dev, gpio_edge_t mode, void (*fptr)(void *), void * args)
```
I see in [mraa_internal_types.h](https://github.com/intel-iot-devkit/mraa/blob/master/include/mraa_internal_types.h) the definition of isr
```c
void (* isr)(void *); /**< the interupt service request */
```
I'm a rooking in C language and don't find the solution to works well.

arfoll, 2015-01-28:
Sorry, the isr call does not yet work in js.

arfoll, 2015-02-18:
You can now do:
```javascript
var m = require('mraa')

function h() {
  console.log("HELLO!!!!")
}

x = new m.Gpio(14)
x.isr(m.EDGE_BOTH, h)
````

closed 2015-02-18 by arfoll

richardfickling, 2015-02-19:
:+1: thanks @arfoll!

commit 008184c3 (2015-02-18): nodejs: Add isr support to node.js

This commit adds isr support to node.js mraa module, it also forces
SWIGJAVASCRIPT to be set at compile time by cmake (SWIG uses SWIGJAVASCRIPT and
not SWIGNODE in it's preprocessor). This uses libuv uv_queue_work to call v8isr
and is all done at a C++ level unlike the python isr, so this reuses the
mraa_gpio_isr call. This closes #110