eclipse-mraa/mraa #112: SPI Interface not fully initialized on mraa_spi_init

michael-ring, 2015-02-01:
I found out that on the Beaglebone I need to first set the spi Interface Mode before the interface works correctly.

There are now two possible ways to solve this, either

initialize spi as master per default on mraa_spi_init 

or

change the examples so that they always include a call to mraa_spi_mode

I can provide patches for both, I have no access to galileo/minnow/nuc board so I cannot test if it hurts to do set a default spi mode on init.

I can provide patches for both versions.

tingleby, 2015-02-01:
Thanks Michael!
Out of the two I think "initialize spi as master per default on mraa_spi_init" would be a better fit.
Please go ahead and provide a patch, I will test it out on at least the galileo/edison.

michael-ring, 2015-02-01:
Will do.

Could you please reject my previous pull request, I would like to create a new pull request and take into account the tips I received for the old one. I found out that with the github UI it is only possible to handle one pull request to one repo, there seem to be workarrounds for this but I think it is easier to trash the hole request and to start over with more atomic patches.

tingleby, 2015-02-01:
Are you talking about #95 ?
If you were to submit the PR from a branch then you can have as many PRs as you feel like. If you take a look at AlexT's #111 it has been submitted from a "feature" branch.
Just found this link to try and provide you a rough guide, http://blog.milkfarmsoft.com/2011/03/on-feature-branches/

michael-ring, 2015-02-01:
Thank you, I will have a look at this

michael-ring, 2015-02-14:
Fixed in pull request #125

arfoll, 2015-02-25:
Closed as this is no longer an issue as I understand?

closed 2015-02-25 by arfoll