eclipse-mraa/mraa #114: [question] Using time-sensitive code

petrklus, 2015-02-03:
Hello!

I am planning to interface with a digital AC dimmer. What is not 100% clear to me, how do I define precise wait times as I would do in arduino code? My goal is to convert my arduino dimmer controller, that works by detecting zero-cross, and opening a TRIAC for defined amount of time:

For example:

    void zero_crosss_int()  // function to be fired at the zero crossing to dim the light
    {
      // Firing angle calculation
      // 50Hz-> 10ms (1/2 Cycle) → (10000us - 10us) / 128 = 78 (Approx)
      // 60Hz-> 8.33ms (1/2 Cycle) → (8333us - 8.33us) / 128 = 65 (Approx)
      int dimtime = 0;
      float propTime = 0;
      if(frq){
        dimtime = (77*dimming);
        propTime = 10.0; 
      }
      else{
        dimtime = (65*dimming);
        propTime = 8.33; 
      }
      delayMicroseconds(dimtime);    // Off cycle
      digitalWrite(AC_LOAD, HIGH);   // triac firing
      delayMicroseconds(propTime);         // triac On propogation delay
      digitalWrite(AC_LOAD, LOW);    // triac Off

    }

My main question is - how do I determine if I am using the main CPU or if the code runs on the Quark? It's 

More details re. the dimmer module can be found here:

http://www.inmojo.com/store/inmojo-market/item/digital-ac-dimmer-module-lite-v.2/

alext-mkrs, 2015-02-04:
I would say this question is better suited for Edison or Galileo community at makers.intel.com, because it has nothing to do with mraa :smiley: 

And looks like you have two questions here:
1. How to add a delay into your program, similar to Arduino's `delayMicroseconds` function. This generally depends on the language you use and don't expect it to be precise at the end of the day. Unless you have an RTOS or a plain microcontroller without the OS overhead and competing tasks - it's hard to manage such short time periods. That's just not the best application for this type of the system;
2. How to distinguish whether your code is running on main CPU or Quark. This one is a bit unclear to me. Are you talking about Edison here?

petrklus, 2015-02-04:
Thank you for your reply! The no. 1 I completely understand, which brings me to to 2. According to Edison spec, it should have: "2 Atom Silvermont cores running at 500 MHz and one Intel Quark core at 100 MHz (for executing RTOS ViperOS)". That's why I was asking about the Quark, as I understand that the normal linux will not be able to guarantee any precise timing. As your library seems to be a de-facto way of interfacing with the pins, I thought that there would be a way to delegate code to run on the Quark and thus maintain realtime-ness.

arfoll, 2015-02-04:
@petrklus whilst the quark core is running viper OS there is still currently no way of using it directly so your code is essentially always running on the atom cores. Sorry!

closed 2015-02-04 by arfoll

petrklus, 2015-02-04:
@arfoll Thanks! Is there any roadmap when and how will the functionality be accessible? I can think of a dozen projects that can benefit from this. Also, how is the Atom maintaining comms with i2c/spi when not really real-time? Is it handled by lower-level drivers?

arfoll, 2015-02-04:
@petrklus sadly nothing I can share here, maker.intel.com is the right place to ask for this info.

Realistically this is the issue with every MCU<->Linux development, if you look at the arduino YUN/lininio architecture they have this exact problem. Writing code on the micro only helps if you do not have to go back to Linux to make a decision on the operation. I2c & SPI are not bitbanged they are handled by real hardware on the SoC.

petrklus, 2015-02-06:
Thanks! Will add arduino to the Edison for the time being and use Edison
just for the web UI..