eclipse-mraa/mraa #117: Initializing Aio instance fails on Galileo Gen 2 (aio: unable to setup multiplexers for pin)

rwaldron, 2015-02-07:
- Galileo Gen 2
- mraa@v0.5.4-110-g459ecc0

Test case: 

```js
var io = require("mraa");

var a0 = new io.Aio(0);
var a1 = new io.Aio(1);
var a2 = new io.Aio(2);
var a3 = new io.Aio(3);
var a4 = new io.Aio(4);
var a5 = new io.Aio(5);
```

Result: 

```
root@quark016c31:~# node aio.js

/home/root/aio.js:7
var a4 = new io.Aio(4);
         ^
Error: Invalid AIO pin specified - do you have an ADC?
    at Object.<anonymous> (/home/root/aio.js:7:10)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:906:3
```

```
root@quark016c31:~# ls /sys/bus/iio/devices/iio:device0/
buffer		   in_voltage2_raw    in_voltage5_raw	 name
dev		   in_voltage2_scale  in_voltage5_scale  power
in_voltage0_raw    in_voltage3_raw    in_voltage6_raw	 scan_elements
in_voltage0_scale  in_voltage3_scale  in_voltage6_scale  subsystem
in_voltage1_raw    in_voltage4_raw    in_voltage7_raw	 trigger
in_voltage1_scale  in_voltage4_scale  in_voltage7_scale  uevent
```

```
root@quark016c31:~# journalctl -f
-- Logs begin at Tue 2014-11-04 00:38:00 UTC. --
Feb 07 17:50:00 quark016c31 kernel: intel_qrk_gip 0000:00:15.2: controller timed out
Feb 07 17:50:00 quark016c31 kernel: pca953x 0-0027: failed writing register
Feb 07 17:50:01 quark016c31 kernel: intel_qrk_gip 0000:00:15.2: controller timed out
Feb 07 17:50:01 quark016c31 kernel: pca953x 0-0027: failed writing register
Feb 07 17:50:02 quark016c31 kernel: intel_qrk_gip 0000:00:15.2: controller timed out
Feb 07 17:50:02 quark016c31 kernel: pca953x 0-0027: failed writing register
Feb 07 17:50:03 quark016c31 kernel: intel_qrk_gip 0000:00:15.2: controller timed out
Feb 07 17:50:04 quark016c31 kernel: intel_qrk_gip 0000:00:15.2: controller timed out
Feb 07 17:50:04 quark016c31 kernel: pca953x 0-0027: failed writing register
Feb 07 17:50:04 quark016c31 libmraa[686]: aio: unable to setup multiplexers for pin
```

tingleby, 2015-02-08:
Do you have, perchance, A4 or A5 connected to anything like GND.

rwaldron, 2015-02-08:
Connected to nothing.

tingleby, 2015-02-08:
Ok, Ill take a look next chance I have.

arfoll, 2015-02-09:
I don't see this behaviour on my gen2. It looks from the log as if your kernel module has gone haywire. weird.

tingleby, 2015-02-10:
I have been unable to reproduce this.

arfoll, 2015-03-19:
Closing due to lack of ability to reproduce, likely hardware issue.

closed 2015-03-19 by arfoll

egradman, 2015-03-24:
I'm having exactly the same issue.  Aio:4

```
Sep 03 12:01:32 quark017953 kernel: intel_qrk_gip 0000:00:15.2: controller timed out
Sep 03 12:01:32 quark017953 kernel: pca953x 0-0027: failed writing register
Sep 03 12:01:33 quark017953 kernel: intel_qrk_gip 0000:00:15.2: controller timed out
Sep 03 12:01:33 quark017953 kernel: pca953x 0-0027: failed writing register
Sep 03 12:01:34 quark017953 kernel: intel_qrk_gip 0000:00:15.2: controller timed out
Sep 03 12:01:35 quark017953 kernel: intel_qrk_gip 0000:00:15.2: controller timed out
Sep 03 12:01:36 quark017953 kernel: intel_qrk_gip 0000:00:15.2: controller timed out
Sep 03 12:01:37 quark017953 libmraa[298]: gpio: Failed to write 78 to export
Sep 03 12:01:37 quark017953 libmraa[298]: aio: unable to setup multiplexers for pin
Sep 03 12:01:37 quark017953 kernel: intel_qrk_gip 0000:00:15.2: controller timed out
```

I'll try with new hardware.

egradman, 2015-03-24:
I swapped out the Galileo (same SD card) and now the problem happens intermittently.

```
> new mraa.Aio(5)
Error: Invalid AIO pin specified - do you have an ADC?
    at repl:1:2
    at REPLServer.self.eval (repl.js:110:21)
    at Interface.<anonymous> (repl.js:239:12)
    at Interface.EventEmitter.emit (events.js:95:17)
    at Interface._onLine (readline.js:202:10)
    at Interface._line (readline.js:531:8)
    at Interface._ttyWrite (readline.js:760:14)
    at ReadStream.onkeypress (readline.js:99:10)
    at ReadStream.EventEmitter.emit (events.js:98:17)
    at emitKey (readline.js:1095:12)
> new mraa.Aio(5)
{}
> new mraa.Aio(5)
{}```

```
Sep 03 11:46:23 quark017953 kernel: intel_qrk_gip 0000:00:15.2: controller timed out
Sep 03 11:46:24 quark017953 kernel: intel_qrk_gip 0000:00:15.2: controller timed out
Sep 03 11:46:24 quark017953 kernel: pca953x 0-0027: failed writing register
```

xbolshe, 2016-03-11:
This issue may be related with this one: https://github.com/intel-iot-devkit/mraa/issues/390
When a pull-up/pull-down resistor is enabled on A4/A5, an I2C bus is blocked with messages like shown above. That causes the issue.