eclipse-mraa/mraa #119: python I2c.write() with invalid arg causes a segfault

arfoll, 2015-02-11:


tripzero, 2015-02-15:
Here's a stack trace and how to reproduce:
```python
import mraa
x = mraa.I2c(0)
x.address(0x40)
x.write(0x04)
```
Program received signal SIGSEGV, Segmentation fault.
0xb7eed7ed in mraa_i2c_write () from /usr/lib/python2.7/site-packages/_mraa.so
(gdb) bt
#0  0xb7eed7ed in mraa_i2c_write () from /usr/lib/python2.7/site-packages/_mraa.so
#1  0xb7eeaec7 in mraa::I2c::write(unsigned char const*, int) () from /usr/lib/python2.7/site-packages/_mraa.so
#2  0xb7ee3f5b in _wrap_I2c_write () from /usr/lib/python2.7/site-packages/_mraa.so
#3  0x4ecf6aaf in PyCFunction_Call () from /usr/lib/libpython2.7.so.1.0
#4  0x4ed565fb in PyEval_EvalFrameEx () from /usr/lib/libpython2.7.so.1.0
#5  0x4ed576c1 in PyEval_EvalCodeEx () from /usr/lib/libpython2.7.so.1.0
#6  0x4ed568c4 in PyEval_EvalFrameEx () from /usr/lib/libpython2.7.so.1.0
#7  0x4ed576c1 in PyEval_EvalCodeEx () from /usr/lib/libpython2.7.so.1.0
#8  0x4ed57843 in PyEval_EvalCode () from /usr/lib/libpython2.7.so.1.0
#9  0x4ed71369 in ?? () from /usr/lib/libpython2.7.so.1.0
#10 0x4ed73536 in PyRun_InteractiveOneFlags () from /usr/lib/libpython2.7.so.1.0
#11 0x4ed73713 in PyRun_InteractiveLoopFlags () from /usr/lib/libpython2.7.so.1.0
#12 0x4ed73d38 in PyRun_AnyFileExFlags () from /usr/lib/libpython2.7.so.1.0
#13 0x4ed837cc in Py_Main () from /usr/lib/libpython2.7.so.1.0
#14 0x08048567 in main ()

arfoll, 2015-02-15:
Yes I should have been clearer. For now use writeByte if your argument isn't a bytearray. If you give write() anything but a bytearray it'll likely segfault instead of simply complaining.

arfoll, 2015-02-25:
Fixed with a2220e36362246cd0c0748924f3731600f73d488

```
>>> import mraa as m
>>> x = m.I2c(0)
>>> x.write(0x00)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "mraa.py", line 612, in write
    return _mraa.I2c_write(self, *args)
ValueError: bytearray expected
```

closed 2015-02-25 by arfoll