eclipse-mraa/mraa #122: PWM config_ms(int period, float duty) needs better documentation and is not doing anything on Galileo v1

cosscat, 2015-02-13:
Hello,

While trying tho get more in debth with PWM on mraa v0.5.4-118-g818e1a8, when issuing (in Pyhton) the help(), the documentation didn't pointed out to when this function should be used (before or after enable(True)). Also, tried it with two different duty cycle values 0.3 and 0.7 and the intensity of the LED is the same in both cases. 
Here is my code:

>>> import mraa as m
>>> import mraa as m
>>> import mraa as m

alext-mkrs, 2015-06-14:
@cosscat the official Python documentation for this method is here: http://iotdk.intel.com/docs/master/mraa/node/classes/pwm.html#method_config_ms. What you see by using "help" is I think autogenerated by SWIG and contains only stuff it could find out automatically, i.e. the class it belongs to and argument types - not much and you better use the recommended docs source for finding information.

As far as the failure is concerned, everything is working correctly actually :smiley:. For Gen1 the max PWM period you can set is just below 8 ms and you've tried to set 10, which is not supported. Return value of "4" you've got means the parameter you've set is invalid, which it is.

The function can be used at any time, as far as I can see from the code. In the worst case you'll get an error code instead of 0, so check for that.

All in all, I think this one could be closed as not an actual bug, @cosscat, @arfoll, @tingleby, what do you think?

arfoll, 2015-06-21:
This should be fixed with the max pwm stuff now

closed 2015-06-21 by arfoll