eclipse-mraa/mraa #128: I2C module accepts addresses that are out of range of the 7 bit standard

cosscat, 2015-02-16:
Hello,
I was doing some tests on the I2c class and observed that when setting the address, I can use also addresses that extend from 7 bits: eg. 0x41. The last bit -LSB- should be reserved for direction on the bus: 0 for write on the bus and 1 for read from the bus. I think an error should be thrown when violating this rule.

Tests done on: 
>>> m.getVersion()
'v0.5.4-118-g818e1a8'

Regards,

Costin

arfoll, 2015-02-17:
This is interesting but I think we probably support 10bit i2c addresses too, which is why the ioctl doesn't fail. but true we don't do any checking maybe we should

arfoll, 2015-02-20:
We do the correct thing AFAIK. We support only 7bit addressing (feature anyone??? :)). You're correct the LSB is the write/read but the kernel does that for us and it's done with the 8th bit not the 7th. So the max valid address is 0x7F and as you can see 0x80 fails. 0x41 is perfectly valid.

```
>>> x.address(0x7F)                                                                                                           
0
>>> x.address(0x80)                                                                                                           
5
```

closed 2015-02-20 by arfoll