eclipse-mraa/mraa #131: Cannot run OnBoard LED Blink example on Galileo Gen 1

abdullahyildiz, 2015-02-17:
I cannot run the default example which uses the GPIO pin 13 for the onboard LED:

```javascript
var mraa = require('mraa'); //require mraa
console.log('MRAA Version: ' + mraa.getVersion()); //write the mraa version to the Intel XDK console
var myOnboardLed = new mraa.Gpio(13); //LED hooked up to digital pin 13 (or built in pin on Galileo Gen1 & Gen2 plus Edison Arduino Breakout board)
myOnboardLed.dir(mraa.DIR_OUT); //set the gpio direction to output
```
When I run the code, the onboard LED does not blink. However, when I use an Arduino LED blink sketch using Arduino IDE (and use the pin 13), it works.

I tried updating mraa and upm, but it didn't work.

As a workaround,

I changed the line 85 at file api/mraa/gpio.hpp:

```c++
Gpio(int pin, bool owner=true, bool raw=false) {
```
as

```c++
Gpio(int pin, bool owner=true, bool raw=true) {
```

to make mraa use `mraa_gpio_init_raw(pin)` function which afterwards accesses sysfs:

```c++
if (raw) {
                m_gpio = mraa_gpio_init_raw(pin);
 }
```

Then built mraa and installed the new node module instead of the existing one. 

Then I changed the GPIO pin number to 3 instead of 13 in the example above:

```javascript
var myOnboardLed = new mraa.Gpio(3);
```

This is because the GP LED (onboard LED) has the GPIO pin number of 3 within sysfs and as I see, mraa uses sysfs to access the GPIO pins.

I know it is not a good solution but I couldn't find an easier way.

arfoll, 2015-02-17:
Did you try this?
`var myOnboardLed = new mraa.Gpio(3, false, true);`

And yes on gen1 there is no good way of doing this as the LED is not on the same pin as pin13, it's an annoying hardware 'feature'.

abdullahyildiz, 2015-02-17:
```javascript
var myOnboardLed = new mraa.Gpio(3, false, true);
```
is definitely a lot easier :+1:

Do you have an idea that why using the pin number 13  on Arduino sketch worked? That is, is it the firmware which is kept on 8 MByte SPI Flash that translates the pin numbers?

arfoll, 2015-02-17:
When you compile an arduino sketch for a g1 the wiring library has a hack to change both gpios at the same time. I don't like this 'feature' so I decided not to implement it as the LED & pin13 won't flip at the same frequency and I consider the behaviour confusing and essentially removes a 'fast' GPIO. Also previously the library couldn't really handle it, nowadays it would be fairly easy with an advance hook. I've just updated our doc to mention it with 78fe6f427082ba4541ae190af2c65da3d531b7a7

closed 2015-02-17 by arfoll