eclipse-mraa/mraa #132: SPI module, setting frequency imposes no limit

cosscat, 2015-02-17:
Hello,

While testing SPI of mraa on Gallileo v1, I observed that there is no limit for setting the frequency in Hz except the boundries of integer values. Is there a way to interogate the hardware for the max. frequency it accepts?

>>> m.getVersion()
'v0.5.4-118-g818e1a8'

>>> s.frequency(1000)
0
>>> s.frequency(1000000)
0
>>> s.frequency(10000000000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/mraa.py", line 795, in frequency
    return _mraa.Spi_frequency(self, *args)
TypeError: in method 'Spi_frequency', argument 2 of type 'int'
>>> 

Regards,

Costin

arfoll, 2015-02-17:
Yes, we probably should have a MAX limit for spi bus freq, we currently don't and just accept anything that spidev will let us set.

michael-ring, 2015-02-17:
I think there are two problems here.

First is that python is called with something that is most likely too big for an int. (No idea if this is really the case, me not python)

the second problem is that the frequency value can be too high for the spi interface, this problem should be fixed with latest commit, you should get a warning in syslog and the maximum is set to what the spi says is max.

raoloka, 2015-06-08:
On the other hand, even though the function is called to increase the frequency, it doesn't seem to be
effective. I noticed the clock frequency still hovered around 4MHz.

alext-mkrs, 2016-02-06:
Looks like we can close this one - the original issue is fixed by @michael-ring in his commit 3a5b99252419b5, so we now check the driver for the max SPI frequency.

We also have a hardcoded value in `mraa_spi_init_raw()`, which caps the frequency to 4MHz, even if driver tells us it can support more. And that magic value was there since the very beginning of the SPI support in mraa, as far as I can see. @tingleby, is there any specific idea behind the value (you were the author of that first commit :smiley:)?

What's interesting is that we don't do the same hardcoding in `mraa_spi_set_frequency()`, we only interrogate the driver - which looks like an inconsistency to me. Should we remove the hardcoded value from `_init_raw()`, maybe?

alext-mkrs, 2016-02-28:
@tingleby, ping - any comments on the above?

tingleby, 2016-03-01:
The cap of 4Mhz was probably a limitation of the original board, Galileo Gen1. If you can get the clock higher than that, then by all means.

alext-mkrs, 2016-03-28:
Ok, thanks - I'll submit a patch shortly then.

commit ecf049ec (2016-04-11): spi.c: removed hard frequency cap in mraa_spi_init_raw()

We rely on driver-supplied information in mraa_spi_frequency(),
so let's be consistent.

Closes #132.
closed 2016-04-11 by arfoll