eclipse-mraa/mraa #138: UART module accepts initialising with the port no of the serial console

cosscat, 2015-02-25:
Hello,

I did this test on Galileo v1 and when initialising Uart with port no. 1 it reports no warning/error message. The /dev/ttyS1 is assigned to the serial console (the on-board jack alike serial port). Also getDevicePath() reports back the assigned port:

>>> import mraa as m, serial as s
>>> uart=m.Uart(1)
>>> uart.getDevicePath()
'/dev/ttyS1'

Costin

arfoll, 2015-02-28:
@cosscat I'm not too sure if we want to fix this, it's nice to be able to use that serial port from mraa if the user wants and at the same time there's no good platform independant way of finding out if a serial port is already in use. We could hack something up to check systemd services/gettys but I think that's horribly nasty.

The best I can think of is to document this in the galileo gen1 platform doc file. What do you think?

cosscat, 2015-05-08:
Hi Brendan, 

Sorry for the delay in getting back to this. Yes, i guess what you said is fine. Adding something to the doc file is best.

alext-mkrs, 2015-06-14:
I've just checked the sources and what's interesting is that for Gen1 we specify two UART devices in the platform config file, while for Gen2 we specify only one (excluding the one with the console). Is there any specific idea behind this? Should we maybe align the two and then document as already agreed in this thread?

tingleby, 2015-06-29:
I went ahead and aligned them, no reason not to.

commit 1ef71bb3 (2015-06-29): x86: Galileo Gen 2: add ftdi header serial

Closes #138
closed 2015-06-29 by tingleby