eclipse-mraa/mraa #139: Undefined reference to 'mraa_spi_write_word' or  'mraa_spi_write_buf_word'

JulioLP, 2015-02-26:
I´m programming accelerometer data receiving via SPI. 

The problem is, even though every other function of the library works fine for me, this two ('mraa_spi_write_word' and 'mraa_spi_write_buf_word') can not be complied in my Edison. The OS is yockto linux (last version) and the version of mraa is 0.6 (also last). I´m using gcc and I am including -lmraa for the compilation.

I´ve also tried compling the example given in the master branch (the one named spi_max7219), but I also can´t get it complied.

I tried to reinstall the libraries, but nothing new happens.

¿Does somebody have an idea about what´s happening?
Thank you all.

oloynet, 2015-02-26:
In MRAA 0.6 there is no method for **mraa_spi_write_word** or **mraa_spi_write_word** like in the master branch
See : https://github.com/intel-iot-devkit/mraa/blob/v0.6.0/src/spi/spi.c

Lke my previous issue #137 for MAX31865, you can try to set SPI to works in 16 bits per word
```c
    // ...... init

    mraa_spi_context spi;
    spi = mraa_spi_init(0);
    mraa_spi_mode(spi, MRAA_SPI_MODE3); // supported mode depend on the MAX7219/MAX7221 specs
    mraa_spi_frequency(spi, 1000000);

    mraa_spi_bit_per_word(spi, 16); // 16 bits mode

    // ...... writing

    uint8_t tx_read[] = {0x01,0x00}; // set yours values (LSB / MSB)
    uint8_t *rx_read;

    rx_read = mraa_spi_write_buf(spi, tx_read, 2); // writing 2 bytes

    printf(" > Write 0x%02x 0x%02x -- Read 0x%02x 0x%02x\n", tx_read[1], tx_read[0],  rx_read[1], rx_read[0]);
```

arfoll, 2015-02-26:
@oloynet is correct, essentially 0.6.0 is behind master (which sadly still  builds as 0.5.4 - don't ask!). I've just tagged/released 0.6.1 which fixes this issue, 0.6.1 has the functions you're looking for. It should be in the intelgalactic opkg repo shortly.

closed 2015-02-26 by arfoll

JulioLP, 2015-02-26:
Thank you both! I tend to get confused when using github, always wrong about the branch I`m really using (also the version numbers here are a bit strangely organised :) ). You saved me a lot of time!

arfoll, 2015-02-26:
Yes I messed up on v0.6.0, in the future the part where the branch branched off will be tagged.