eclipse-mraa/mraa #142: getPlatformName() returns "Intel Galileo Gen 1" on haswell

tripzero, 2015-03-01:
Running mraa on my haswell-based desktop, I got "Intel Galileo Gen 1" when querying getPlatformName().

tingleby, 2015-03-01:
Yes, this is the current expected behaviour if mraa is running on a platform is is not equipped to work with. If you check your syslog, it should be complained a plenty in there.

tripzero, 2015-03-01:
Shouldn't it return ""?  Or "unknown system"?

tripzero, 2015-03-01:
Let me describe the use-case that's broken here:

Let's say you want to write a higher-level library that introspects mraa using getPlatformName and uses pre-defined pin settings for that platform.

If an unknown system returns "Intel Galileo Gen 1", then the pin-outs will be wrong and the library will be broken

tingleby, 2015-03-01:
This is an element of legacy behaviour, nerveless something that could do with a rethink.

My current proposal is: When the library is run on a system without additional definitions it will attempt to find the limitations of the current system, and in the case of getPlatformName return something like unknown.

arfoll, 2015-03-01:
Agreed this behaviour was pretty rubbish, it makes no sense now that we support quite a few platforms, fixed & changed. Note that arm module by default will init an rpi as well. This is a bit more difficult to treat since arm platforms rarely have a nicely exported name such as the one in DMI.

closed 2015-03-01 by arfoll

tingleby, 2015-03-01:
Now the behaviour is
```
mraa: FATAL error, failed to initialise platform
hello mraa
 Version: v0.6.1-8-g2c74ba7
 Running on (null)
```
@arfoll I'll talk to you later about my further plans for this.

commit 10f450b3 (2015-03-01): x86.c: unknown init now returns MRAA_UNKNOWN_PLATFORM

This stops the legacy beahviour that was to initialise as a gen1 galileo if we
failed to find a valid dmi name. Closes #142