eclipse-mraa/mraa #143: i can't install mraa at intel edison, how can i ??

digerati73, 2015-03-01:
below text is copied from putty.

i want to do this
echo "src maa-upm http://iotdk.intel.com/repos/1.1/intelgalactic" > /etc/opkg/intel-iotdk.conf
opkg update
opkg upgrade

but i failed...

/sbin/ldconfig: file /usr/lib/libfaad.so.2 is truncated

/sbin/ldconfig: file /usr/lib/libfaad.so.2.0.0 is truncated

/sbin/ldconfig: Writing of cache data failed: No space left on device

Configuring libfreetype6.
/sbin/ldconfig: file /usr/lib/libfaad.so.2 is truncated

/sbin/ldconfig: file /usr/lib/libfaad.so.2.0.0 is truncated

/sbin/ldconfig: Writing of cache data failed: No space left on device

Collected errors:
 * parse_from_stream_nomalloc: Missing new line character at end of file!
 * parse_from_stream_nomalloc: Missing new line character at end of file!
 * parse_from_stream_nomalloc: Missing new line character at end of file!
 * copy_file_chunk: write: No space left on device.
 * copy_file_chunk: write: No space left on device.
 * copy_file_chunk: write: No space left on device.
 * copy_file_chunk: write: No space left on device.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/arch/x86/kernel/: No space left on device.
 * wfopen: /lib/modules/3.10.17-yocto-standard/kernel/arch/x86/kernel/test_nx.ko: No such file or directory.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/net/: No space left on device.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/net/netfilter/: No such file or directory.
 * wfopen: /lib/modules/3.10.17-yocto-standard/kernel/net/netfilter/nf_conntrack.ko: No such file or directory.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/net/: No space left on device.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/net/netfilter/: No such file or directory.
 * wfopen: /lib/modules/3.10.17-yocto-standard/kernel/net/netfilter/nf_conntrack_sane.ko: No such file or directory.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/net/: No space left on device.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/net/netfilter/: No such file or directory.
 * wfopen: /lib/modules/3.10.17-yocto-standard/kernel/net/netfilter/nf_conntrack_proto_gre.ko: No such file or directory.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/net/: No space left on device.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/net/netfilter/: No such file or directory.
 * wfopen: /lib/modules/3.10.17-yocto-standard/kernel/net/netfilter/nf_conntrack_proto_udplite.ko: No such file or directory.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/net/: No space left on device.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/net/netfilter/: No such file or directory.
 * wfopen: /lib/modules/3.10.17-yocto-standard/kernel/net/netfilter/nf_nat.ko: No such file or directory.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/net/: No space left on device.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/net/netfilter/: No such file or directory.
 * wfopen: /lib/modules/3.10.17-yocto-standard/kernel/net/netfilter/xt_sctp.ko: No such file or directory.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/drivers/video/: No space left on device.
 * wfopen: /lib/modules/3.10.17-yocto-standard/kernel/drivers/video/sysimgblt.ko: No such file or directory.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/drivers/video/: No space left on device.
 * wfopen: /lib/modules/3.10.17-yocto-standard/kernel/drivers/video/syscopyarea.ko: No such file or directory.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/drivers/video/: No space left on device.
 * wfopen: /lib/modules/3.10.17-yocto-standard/kernel/drivers/video/sysfillrect.ko: No such file or directory.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/drivers/gpu/: No space left on device.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/drivers/gpu/drm/: No such file or directory.
 * wfopen: /lib/modules/3.10.17-yocto-standard/kernel/drivers/gpu/drm/drm_kms_helper.ko: No such file or directory.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/drivers/gpu/: No space left on device.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/drivers/gpu/drm/: No such file or directory.
 * wfopen: /lib/modules/3.10.17-yocto-standard/kernel/drivers/gpu/drm/drm_usb.ko: No such file or directory.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/drivers/gpu/: No space left on device.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/drivers/gpu/drm/: No such file or directory.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/drivers/gpu/drm/udl/: No such file or directory.
 * wfopen: /lib/modules/3.10.17-yocto-standard/kernel/drivers/gpu/drm/udl/udl.ko: No such file or directory.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/net/: No space left on device.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/net/netfilter/: No such file or directory.
 * wfopen: /lib/modules/3.10.17-yocto-standard/kernel/net/netfilter/xt_helper.ko: No such file or directory.
 * extract_archive: Cannot make dir /lib/modules/3.10.17-yocto-standard/kernel/net/: No space left on device.
[... 1098 lines cut ...]
 * copy_file_chunk: write: No space left on device.
 * copy_file_chunk: write: No space left on device.
 * copy_file_chunk: write: No space left on device.
 * copy_file_chunk: write: No space left on device.
 * copy_file_chunk: write: No space left on device.
 * copy_file_chunk: write: No space left on device.
 * copy_file_chunk: write: No space left on device.
 * copy_file_chunk: write: No space left on device.
 * copy_file_chunk: write: No space left on device.
 * copy_file_chunk: write: No space left on device.
 * copy_file_chunk: write: No space left on device.
 * pkg_run_script: package "libtdb1" postinst script returned status 1.
 * opkg_configure: libtdb1.postinst returned 1.
 * pkg_run_script: package "libfreetype6" postinst script returned status 1.
 * opkg_configure: libfreetype6.postinst returned 1.
 * pkg_run_script: package "libfaad2" postinst script returned status 1.
 * opkg_configure: libfaad2.postinst returned 1.
 * pkg_run_script: package "libwbclient0" postinst script returned status 1.
 * opkg_configure: libwbclient0.postinst returned 1.
 * pkg_run_script: package "upm" postinst script returned status 1.
 * opkg_configure: upm.postinst returned 1.
 * pkg_run_script: package "samba" postinst script returned status 1.
 * opkg_configure: samba.postinst returned 1.
 * opkg_conf_write_status_files: Couldn't close //var/lib/opkg/status: No space left on device.
**********************************************************************************

digerati73, 2015-03-01:
what is the maximum size of SDHC card ?
i am using 8G size
[image]
[image]

michael-ring, 2015-03-01:
look at the size of your / filesystem, it is 100% full.

I looks that your last install  (ffmpeg or something that uses ffmpeg I guess) alreay failed as I see that /usr/lib/libfaad.so.2 is truncated.

You need to free up some space on root filesystem to be able to install anything on your edison.

arfoll, 2015-03-01:
@digerati73 having a SDHC card makes no difference see it's mounted on /media/sdcard but you've run out of space in /. Try the latest r2 edison images they use AUFS to share / & /home so you might not run out of space so quick.

closed 2015-03-01 by arfoll