eclipse-mraa/mraa #144: [nodejs] building with swig 3.0.5 and nodejs 0.12.0 is broken

arfoll, 2015-03-01:
Tested on arch linux,

arfoll, 2015-03-03:
After a bit of hunting we now need to define SWIG_V8_VERSION  to 0x032873 (nodejs 0.12.0 platform) in the SWIG generated code. This fixes a multitude of issues but we still have some issues with v8::Undefined for example. We also now need to include <uv.h> in gpio.hpp header to get uv_work_t and similar.

Here is the issue http://dpaste.com/0XPA6VW if someone wants to hunt.

arfoll, 2015-03-03:
I guess all the answers are now in here:
https://strongloop.com/strongblog/node-js-v0-12-c-apis-breaking/

This requires a bit of rethink do we care about nodejs 0.12 yet?

rwaldron, 2015-03-03:
> This requires a bit of rethink do we care about nodejs 0.12 yet?

I would say yes, but to buy time you might consider simply failing with a message that says something like "currently only supports node 0.10.x". That would at least make it clear to the end developer that node 0.12.x isn't going to be successful yet, and how to get around that.

arfoll, 2015-03-10:
I got it building with 0.12 in this branch https://github.com/intel-iot-devkit/mraa/tree/node012. Unclear whether ISRs work. Note if someone wants to test this you need to set `#define SWIG_V8_VERSION 0x032873`. I need to find a good way of setting this from cmake/binding.gyp, node seems to export `NODE_MODULE_VERSION` and 0x000D (v0.11.13+) seem to be what we want to aim for.

arfoll, 2015-03-18:
Updated branch builds for node 0.10.x and 0.12.x but now depends on SWIG 3.0.5. NPM pkg could work on both but would need to be updated.

arfoll, 2015-03-18:
Does anyone know how to get the nodejs (or even v8 version) from binding.gyp at compile time?

arfoll, 2015-03-19:
Now merged. Npm pkg is still not updated because our build machine is using swig 3.0.3 - @tingleby can you take care of that?

Note that from now on swig 3.0.5 is required to build mraa nodejs binding.

tingleby, 2015-03-20:
Build Machine updated. Builds passing!

closed 2015-05-20 by arfoll