eclipse-mraa/mraa #146: Error Code gets lost in mraa_spi_write

michael-ring, 2015-03-01:
I am right now extending the spi module to learn new tricks and during that process I saw a problem in the code of mraa_spi_write:

This function always returns an uint8_t. when for some reason the ioctl for sending the data fails it returns -1 (=0xff), so an error situation cannot get distinguished from a proper return of 0xff

Any ideas?

arfoll, 2015-03-01:
This is definately not ideal, -1 definately is not the right thing to return and I'm surprised our coverity scans haven't picked that up...

I dont know of a good solution but I think we can do two things, in cases of error complain in syslog and return 0 or the value that was written. I'm edging towards 0 being a fail. I'd really rather avoid changing the function signature, I guess we could also set errno so that people dead set on checking the write success can do so.

michael-ring, 2015-03-01:
I was hoping on a signature change (returning an int instead of uint_8 but syslog & returning 0 if fine.
errno is perhaps not so ideal because it is used nowhere else in mraa (At least I never stumbled over this)

arfoll, 2015-03-01:
Agreed, I'd like to avoid errno because it makes the lib more complex than
is like.

Let me sleep on the signature change its probably relatively harmless.

commit f854a2b4 (2015-03-03): spi: return int instead of uint8_t in mraa_spi_write

This stops the error code getting lost since -1 wraps around to 0xFF when cast
to a uint8_t. This fixes #146
closed 2015-03-03 by arfoll