eclipse-mraa/mraa #147: SPI transfer with 16 bits per word for platforms that do not support this in hardware

michael-ring, 2015-03-01:
I have implemented and tested code to enable support for 16bit transfers with the Raspberry Pi. There is still support for some cases missing (lsb first and MODE1) but so far this works really nice.
I would like to add checks to spi_write and spi_write_word that they refuse to send data when the numer of bits set does not match the size of the data. example:

mraa_spi_write should refuse to work when bits_per_word is >8
and
mraa_spi_write_word should refuse to work when bits per word is <9 and >16

Not doing so will create quite some special cases that need to get handled and code would be cleaner as cases like a user wanting to send 3 bytes in 16bit mode simply would not exist.

Are you guys OK with those changes?

Michael

arfoll, 2015-03-01:
Do you mean: "mraa_spi_write_word should refuse to work when bits per word is *not* 16" or am I missing something?

I'm happy with something like this as long as there is good syslog(LOG_ERR, "") message on those cases which is clear. If there are specific rpi type things you are worried about then it's perfectly fine to have advance functions specific to that platform to stop 'dangerous' cases. One fo things I've been thinking about for I2c is to probe the i2c-dev funcs before doing things, sadly I don't think there is such a think for spidev.

michael-ring, 2015-03-01:
Yes, this is one of the cases I wanted to cover.

I also would like the same to be true for mraa_spi_write (it should not try to transfer 8 bits when bits_per_word is set to 16 because the clock will always do 16 cycles this means you must transfer 16 bits.

I can promise tons of syslog messages, they are already there ;-) ;-)