eclipse-mraa/mraa #148: Building From NPM fails

srlmproductions, 2015-03-02:
I've tested this with Node v0.10.36 on Ubilinux on the Edison, and Node v0.10.33 on Ubuntu 14.04 (on my desktop).

The package from NPM fails with:
```
...
../src/mraa.c:230:13: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  CC(target) Release/obj.target/mraa/src/version.o
  CXX(target) Release/obj.target/mraa/src/mraajsJAVASCRIPT_wrap.o
../src/mraajsJAVASCRIPT_wrap.cxx:156:23: fatal error: node/node.h: No such file or directory
compilation terminated.
make: *** [Release/obj.target/mraa/src/mraajsJAVASCRIPT_wrap.o] Error 1
make: Leaving directory `/home/pilothouse/pilothouse/node_modules/mraa/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:820:12)
gyp ERR! System Linux 3.10.17-poky-edison-ww42+
gyp ERR! command "node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pilothouse/pilothouse/node_modules/mraa
gyp ERR! node -v v0.10.36
gyp ERR! node-gyp -v v1.0.1
gyp ERR! not ok 

npm ERR! mraa@0.6.1-10-gf33880c install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the mraa@0.6.1-10-gf33880c install script.
npm ERR! This is most likely a problem with the mraa package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls mraa
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 3.10.17-poky-edison-ww42+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! cwd /home/pilothouse/pilothouse
npm ERR! node -v v0.10.36
npm ERR! npm -v 1.4.28
npm ERR! code ELIFECYCLE
npm ERR! not ok code 0
```

The key line is
```
../src/mraajsJAVASCRIPT_wrap.cxx:156:23: fatal error: node/node.h: No such file or directory
```

The following hack will allow mraa to cleanly compile:
```
# Download mraa, but don't build it
npm install --ignore-scripts
cd node_modules/mraa
nano src/mraajsJAVASCRIPT_wrap.cxx
# TODO: Change line 156 from #include <node/node.h> to #include <node.h>
npm install
```

The problem seems to stem from the fact that mraa expects node-dev (which doesn't actually seem to be used any more?) to have put sources in `/usr/include/node`, but recent versions seem to include the headers more dynamically.

A brittle solution that doesn't involve modifying mraa is to put a symbolic link in `/usr/include`:

```
sudo ln -s /home/USERNAME/.node-gyp/0.10.36/src/ /usr/include/node
```

It looks like the first line of `./src/javascript/CMakeLists.txt`  needs to be changed, since that's the only place in the project than mentions `node.h`

arfoll, 2015-03-03:
This is a problem in joyent's nodejs repository, they expect all headers to come from node-gyp instead of the nodejs package itself and they don't build a real nodejs-dev package (I did tweet to them about it but they seemed to have misunderstood the issue).

Here's how we find node.h
```
find_path (NODE_ROOT_DIR "node/node.h" "src/node.h" PATHS /usr/include/nodejs /usr/local/include/nodejs)
```
We're pretty nice, we'll even go hunt in /usr/local!

If you look at the content of ubuntu's 14.04 nodejs package you'll see the headers are all in the right place: http://packages.ubuntu.com/trusty/amd64/nodejs-dev/filelist

It really is that joyent's repo doesn't follow the same structure as every other distro, my recommendantion is to not use it and compile your own node pkg. Obviously your simlink works too but bar searching the whole user's home for .node-gyp I don't really see how to solve this nicely. I'm open to ideas on how to fix this, maybe binding.gyp provides a method for hunting these headers in the .node-gyp folder.

srlmproductions, 2015-03-10:
For any who come across this thread, building from NPM works best. Note that at this time you'll need to build 0.10.x since 0.12.x mraa support is still in progress.

```
git clone https://github.com/joyent/node.git
cd node
git checkout v0.10.36
./configure && make -j 3 && sudo make install
```

Other than taking about an hour, building from source works well. mraa compiles cleanly.

arfoll, 2015-03-10:
Thanks for coming back to me on this - see #144 for node.js 0.12 support.

closed 2015-03-10 by arfoll

MuhammadAnnaqeeb, 2016-02-19:
By chance, I have found a workaround to install mraa on ubuntu 14.04

    sudo apt-get install nodejs-legacy
    npm install mraa

works just fine for mraa 0.9.3 and nodejs 0.10.25

arfoll, 2016-02-23:
To be honest the issue is because ubuntu nodejs packages are broken, we can build on node4/node5 and node 0.12 because it's what travis does :)