eclipse-mraa/mraa #150: [Edison Arduino] Illegal arguments for construction of _exports_Pwm

X-Y, 2015-03-05:
A related discussion here: https://communities.intel.com/thread/60203

For some reasons it's impossible to use any of the pins as PWM, trying to use servo/pwm will leads to the same error message. However, the same code runs fine a day after. 

root@edison:~# node
> var m = require("mraa")
undefined
> var p = new m.Pwm(3)
Error: Illegal arguments for construction of _exports_Pwm
    at repl:1:9
    at REPLServer.self.eval (repl.js:110:21)
    at repl.js:249:20
    at REPLServer.self.eval (repl.js:122:7)
    at Interface.<anonymous> (repl.js:239:12)
    at Interface.EventEmitter.emit (events.js:95:17)
    at Interface._onLine (readline.js:202:10)
    at Interface._line (readline.js:531:8)
    at Interface._ttyWrite (readline.js:760:14)
    at ReadStream.onkeypress (readline.js:99:10)
>

arfoll, 2015-03-05:
this works here, could you do a m.getVersion() to give me an idea of the version/platform you are using?

X-Y, 2015-03-05:
It's v0.6.1
Now the problem happens when I'm using the Edison mini-breakout board. The same Edison and code would work on a Arduino extension board(which had the same problem previously, but disappeared after night). I've tried with 2 mini-breakouts.

arfoll, 2015-03-05:
So on a minibreakout board pin 3 does nothing, it's J17-4, see docs here http://iotdk.intel.com/docs/master/mraa/edison.html - you can also use the gpio.c sample from the examples to list all the pins and their capabilities, that can be helpful.

X-Y, 2015-03-06:
OK thanks! It's quite confusing for me though, when the same pin number refers to different hardware pins when using different platforms. It would also be better for hardware migration, if I don't have to change the values in my code. 

Now it makes sense for the issue on Arduino module. It's possible that I deleted some journey files(to free up space, a bug in Edison makes it so each time one re-flashes the system the journey becomes larger, ending up to hundreds of Mbs) making Edison forgot it was on the Arduino module.

arfoll, 2015-03-06:
Yes, it's not as simple as I'd like, however there's not really a good way of doing this, this is why there is getPlatformType and you can change your pins depending on your hardware.

If you look at types.h you can use the MRAA_INTEL_EDISON_GPxx enum which means you could use the kernel GPIO numbering (or the MRAA_INTEL_EDISON_MINIBOARD_Jxx_yy miniboard enum if you wished).

closed 2015-03-06 by arfoll