eclipse-mraa/mraa #152: mraa_uart_init(int index) -  on Edison will init UART0 regardless of index...

KurtE, 2015-03-06:
Looking at the function: assuming mraa was init, it will always call: mraa_intel_edison_uart_init_pre(int index)

This function, does not even look at the index, but blindly starts the process of initializing the hardware uart. 

Couple possible fixes.  Have the uart_init_pre function check the index.  Or have teh mraa_uart_init function, verify the index is at least valid before calling the init pre...

KurtE, 2015-03-06:
Edit: Actually wondering about the stuff in the Pre as it look like the main function, then tries to look up the pin mapping and set the pin mux for the input and output pin.  

Note: I think there is also another bug in the main function (mraa_uart_init)
    if (pos >= 0) {
        pos = plat->uart_dev[index].tx;
        if (plat->pins[pos].uart.mux_total > 0) {
           ...
Should be:
    pos = plat->uart_dev[index].tx;
    if (pos >= 0) {
        if (plat->pins[pos].uart.mux_total > 0) {

alext-mkrs, 2015-06-14:
@KurtE as far as the index is concerned - it's actually ok, because the pre function is a hook one and it is called from a platfom-specific file, so it's allowed to have intimate knowledge about platfom HW and not to use all the passed parameters. In this specific case index is irrelevant because we have only one UART listed in the [platform config](https://github.com/intel-iot-devkit/mraa/blob/master/src/x86/intel_edison_fab_c.c#L1076) for Edison.

As far as your second note is concerned (regarding that `pos` variable), that looks either like an assumption that if the RX part is not configured properly (i.e. < 0), then it makes no sense to look into the TX part (though then it IMHO asks to be combined with the RX `if` statement instead of creating a separate one), or it could be a straight bug, with a fix like you've proposed.

@tingleby, Tom, looks like you're the original author of this piece of code, could you please comment which one it is? I can prepare a fix based on that.

alext-mkrs, 2015-06-29:
@tingleby, ping :smiley:

tingleby, 2015-06-29:
The hook as the said signature to comply with the hook definition in the internals, Yes it should check for a valid index sorted with 79d1acd17f89a3cecd6bd670eec69ec16b2b6f32
As for your second part that was a straight up bug. Resolved with fd50ea087b658d68fdd61921b8302f9a7f10242a
Thanks again @KurtE and @alext-mkrs

closed 2015-06-29 by tingleby