eclipse-mraa/mraa #154: Feature Request: 1-wire support

jmealo, 2015-03-08:
Please add 1-wire support to mraa.

It looks like it should be supported in Yocto.

Does anyone have any progress on this?

arfoll, 2015-03-09:
You'll have to be more specific, the problem with one wire is one it's reliance on exact timings which from userspace linux is 'hard'. The other issue is that it's heavily board dependant and alot of sensors handle it differently. Hacking it as your [1] shows is trivial from mraa.

It would be interesting to have a look at how to use it from the kernel and expose that from mraa, the 1wire driver seems to however use device tree which galileo/edison does not so this would require a bit of thinking to make it work on intel boards.

akulo, 2015-08-25:
how about DS18B20 support?

wawrow, 2015-08-25:
I've tried using ds18b20 with edison but it seems like it wasn't switching in/out on gpio fast enough for 1wire, could probably be done with two gpios, but i haven't got a chance to try it out.

akulo, 2015-08-25:
the idea is to have it somehow exposed to node.js to consume without knowing all the implementation details

mulderp, 2015-10-25:
A nice example of 1-wire protocol might be the DHT-22: http://www.adafruit.com/datasheets/DHT22.pdf - the DHT series is quite popular in Arduino examples (it's very cheap), e.g. see https://www.adafruit.com/products/386

mulderp, 2015-10-25:
sorry, dht-xx is not one wire: http://forum.arduino.cc/index.php?topic=253899.0

alext-mkrs, 2016-05-09:
Support for 1-wire via UART was merged with a379eb7bf66a7a31a3b95b9bf239dffaaec241a6 and I think it's as close as we can get to it, given the OS we have on board.

@arfoll, do you think we can close this one as done? I would think so.

arfoll, 2016-05-09:
Indeed, completely forgot this bug :) Closed!

closed 2016-05-09 by arfoll