eclipse-mraa/mraa #156: Move platform types from types.h to their own headers

eschaton, 2015-03-09:
There are a bunch of platform-specific types in `<mraa/types.h>` that should probably be in separate headers specific to the platform:

- `maa_intel_edison_miniboard_t` and `mraa_intel_edison_t` should be in a separate `intel_edison.h` header
- `mraa_raspberry_wiring_t` should be in a separate `raspberry_pi.h` header

arfoll, 2015-03-09:
I don't see the point of doing this, these are part of the API so everyone needs them, I'm happy with them in types.h, until that file grows too big.

If you want to send a PR for this I'm happy to merge it.

closed 2015-03-09 by arfoll

eschaton, 2015-03-09:
I cannot submit a PR for making a change like that without going through a lot of red tape. The point of filing this was that everyone doesn't need the definitions for all of these different platforms: Someone using libmraa on RPi doesn't need Galileo definitions, etc.