eclipse-mraa/mraa #157: Move mraa_i2c_mode_t from types.h to i2c.h

eschaton, 2015-03-09:
The header `<mraa/types.h>` purports to be for basic shared types, but it contains an i2c-specific type, `mraa_i2c_mode_t`. That should really be in `<mraa/i2c.h>` instead.

arfoll, 2015-03-09:
Agreed, the struct should 'ideally' be duplicated in the C++ header and the C header so that SWIG uses the correct name. The reason for throwing it in the types.h is that swig takes that file (but not i2c.h) and will strip the prefix away (since we have the namespace/module/package name). So it really depends on whether you prefer enum duplication or things in the right place :) I'm happy to take a PR to change this.

eschaton, 2015-03-09:
I can't submit a PR, sorry.