eclipse-mraa/mraa #158: The mraa_platform_t type should be a struct or string, not an enum

eschaton, 2015-03-09:
The type `mraa_platform_t` that defines a platform should be designed for extensibility, by being a struct that can be defined in a per-platform header, or by being a string that can be managed via a registry within the library. Right now it's an enum, which requires recompilation of the library to extend for a new platform, can cause binary-incompatibility between builds of the library (if new cases are added within the enum), and can cause conflicts between forks of the library (if multiple forks add a new case to the enum with the same value for different purposes).

For example, I might have a Numato Labs USB GPIO device, and I might consider extending `libmraa` to work with it. However, in addition to implementing that support in general, I'd also have to add an entry to `mraa_platform_t` for it. If I had done so prior to Raspberry Pi support, I might have decided my device is element 5 — and then Raspberry Pi would have also been added at element 5, causing a conflict.

tingleby, 2015-03-09:
Yes, I think in principle this isn't a bad idea. Some care will need to be taken as to not break any existing usages of the API.some negative feelings.
As platforms should now already define a string within there board configuration, I see less of a need for this.
Perhaps @arfoll will have stronger feelings for keeping this practice around.

eschaton, 2015-03-09:
Even if the existing API needs to be kept, it (and the enum) can be marked `__attribute__((deprecated))` (for compilers that support it, eg modern cc and clang) and implemented as indices into the registration table for binary compatibility. Existing code would still work, new code would be discouraged from using the old API, and new platforms wouldn't add to the problem by defining their own enum value.

arfoll, 2015-03-09:
My suggestion would be to keep mraa_platform_t as it is, custom platforms would regsiter as MRAA_CUSTOM_PLATFORM and then use the platform name to indentify themselves. It's also a good way to check we're running on something user defined.

I dont see the binary incompatibility argument, if you run code that uses the symbol MRAA_MY_PLAT and your lib doesn't have it the fact you crash is IMHO not an issue.

eschaton, 2015-03-09:
Since these are enums right now, there's a constant generated in the binary that invokes the library, not a symbol that's looked up by the dynamic loader.