eclipse-mraa/mraa #159: Make platforms extensible without modifying libmraa

eschaton, 2015-03-09:
The `libmraa` API should have an `<mraa/platform.h>` header that defines the interface between `libmraa` and an MRAA-supporting platform. The built-in Edison, Galileo, Raspberry Pi, *et al* platforms should be defined in terms of this API, and it should also be possible to register a platform — in the form of a C struct containing a version number, a string name, and a series of callback function pointers — entirely at runtime.

This would allow one to more easily add support to `libmraa` for platforms that aren't already anticipated in its current design, and to maintain different platforms without making as many modifications to the `libmraa` core code.

tingleby, 2015-03-09:
This is certainly the direction I want to see mraa heading towards. @arfoll  and I have skirted around this area of discussion last week.
I also see your #158 Which I would think would be resolved in the growth of this sort of addition to the library. 
Over the next few weeks Ill roll out some implementation ideas into a branch.

eschaton, 2015-03-09:
Awesome, thanks! My expectation for such an API would be something like

    struct _mraa_platform_desc {
        uint32_t version; /*!<< structure version, pass MRAA_PLATFORM_DESC_VERSION_CURRENT */
        constant char *name; /*!<< platform name */
        // callback function pointers for init, deinit, subsystems, etc.
        // or pointers to separate structure for that stuff
    };
    typedef struct _mraa_platform_desc *mraa_platform_desc_t;
    
    mraa_result_t mraa_register_platform(mraa_platform_desc_t platform);

And then I'd call `mraa_register_platform()` before `mraa_init()` in an app that uses its own platform.

arfoll, 2017-02-07:
Whilst far from perfect the JSON API does most of this.

closed 2017-02-07 by arfoll