eclipse-mraa/mraa #160: Node: I2C readWordReg is too quick

srlmproductions, 2015-03-10:
It appears that when I invoke the `readWordReg()` function the read command appears immediately after the write command, with no delay. This is too fast for my I2C peripheral.

If I artificially recreate the same sequence that `readWordReg()` produces, but with a delay between the write and read portions, then the process works.

Here's my Node code:

```
(function () {
    function char(x) {
        return parseInt(x, 16);
    }

    var wind = new mraa.I2c(1);
    wind.address(2);

    setInterval(function () {
        // Unsuccessful
        //console.log(wind.readWordReg(char('0x32')));
        
        // Manual (successful)
        wind.writeByte(char('0x35'));
        var t = wind.read(2);
        console.log(t.readInt16LE(0));

    }, 1000);
})();
```

`readWordReg()` produces the following sequence. Notice the lack of a gap, and the NACK from the slave:
[image]

The manual sequence works, however:
[image]

So, would it be possible to include a delay parameter to the `readWordReg()` function that delays for a minimum number of microseconds?

arfoll, 2015-03-10:
The reason you're not seeing a gap is that this is using `I2C_SMBUS_READ` which is only one command, this is very efficient and the right thing for most devices. In your case I would say doing it manually is currently the way I'd prefer to handle this. I guess we could maybe put an optional delay parameter but the problem is that delays from userspace are only minimum values and very hard to enforce.

srlmproductions, 2015-03-10:
Thanks. I'll go ahead and make sure to do it manually for the time being.

On a related note, I posted this issue in part for the idea, and in part for the documentation. I'd like to help on the documentation in a more formal matter as I learn about mraa. Maybe I can contribute to some sort of FAQ or troubleshooting guide?

arfoll, 2015-03-12:
Definately doc is more than welcome! Have a look at the doxygen stuff we build from here:
https://github.com/intel-iot-devkit/mraa/blob/master/api/mraa/i2c.hpp#L35
which ends up:
http://iotdk.intel.com/docs/master/mraa/classmraa_1_1_i2c.html

In the short term future we'll have jsdoc that gets build from the C++ stuff so if you doc stuff in there it'll reflect in the jsdoc. Platforms also have doc pages here: http://iotdk.intel.com/docs/master/mraa/pages.html

closed 2015-03-12 by arfoll