eclipse-mraa/mraa #161: Node: segfault when using I2C and ISR

srlmproductions, 2015-03-10:
I've run across a curious combination that produces a segfault:

```javascript
'use strict';

var SegfaultHandler = require('segfault-handler');
SegfaultHandler.registerHandler();

var mraa = require('mraa'); //require mraa
console.log('MRAA Version: ' + mraa.getVersion()); //write the mraa version to the console

var button = new mraa.Gpio(36); // GP14
button.dir(mraa.DIR_IN);

var wind = new mraa.I2c(1);
wind.address(2);

var counter = 0;
setInterval(function () {
    wind.read(2);
    console.log(counter ++);
}, 10);
```

The important combination are the `wind.read(2)` line and the two `button` lines. When I run this on a Sparkfun Edison stack, with no button or I2C (at address 2) hooked up I get the following error:

```
PID 3730 received SIGSEGV for address: 0xffffffff
```

If I hook up the I2C slave at address 2 I get:

```
PID 3707 received SIGSEGV for address: 0xe
/home/pilothouse/pilothouse/node_modules/segfault-handler/build/Release/segfault-handler.node(+0xd7d)[0xb76f2d7d]
[0xb76f740c]
node(_ZN2v88internal13GlobalHandles6CreateEPNS0_6ObjectE+0x40)[0x838e9f0]
node(_ZN2v82V818GlobalizeReferenceEPPNS_8internal6ObjectE+0x68)[0x830c4a8]
node(_ZN4node6BufferC1EN2v86HandleINS1_6ObjectEEEj+0x40)[0x8597b30]
node(_ZN4node6Buffer3NewERKN2v89ArgumentsE+0x1a6)[0x8597d86]
node[0x833f21b]
[0x5b40a376]
[0x5b4217cf]
[0x5b42191a]
[0x5b41316a]
node[0x836bb61]
node(_ZN2v88internal9Execution3NewENS0_6HandleINS0_10JSFunctionEEEiPNS2_INS0_6ObjectEEEPb+0x75)[0x836be95]
node(_ZNK2v88Function11NewInstanceEiPNS_6HandleINS_5ValueEEE+0x111)[0x8317f91]
node(_ZN4node6Buffer3NewEPKcj+0x61)[0x8597f41]
/home/pilothouse/pilothouse/node_modules/mraa/build/Release/mraa.node(_Z14_wrap_I2c_readRKN2v89ArgumentsE+0x132)[0xb73de502]
[0x33e126ef]
[0x33e18fa3]
[0x5b424a85]
[0x33e185aa]
[0x33e1130c]
[0x5b40e501]
[0x5b421892]
[0x5b41308a]
node[0x836bb61]
node(_ZN2v88internal9Execution4CallENS0_6HandleINS0_6ObjectEEES4_iPS4_Pbb+0xd0)[0x836d320]
node(_ZN2v88Function4CallENS_6HandleINS_6ObjectEEEiPNS1_INS_5ValueEEE+0x15c)[0x832227c]
node(_ZN4node12MakeCallbackEN2v86HandleINS0_6ObjectEEENS1_INS0_8FunctionEEEiPNS1_INS0_5ValueEEE+0x5e)[0x859460e]
node(_ZN4node12MakeCallbackEN2v86HandleINS0_6ObjectEEENS1_INS0_6StringEEEiPNS1_INS0_5ValueEEE+0x60)[0x85947f0]
node(_ZN4node9TimerWrap9OnTimeoutEP10uv_timer_si+0x66)[0x85bd026]
node(uv__run_timers+0x70)[0x8603e50]
node(uv_run+0x82)[0x85f9852]
```

If it helps, the I2C slave is an Arduino running this code: https://github.com/srlmproductions/pilothouse/blob/master/ma3toI2C/ma3toI2C.ino

srlmproductions, 2015-03-12:
I got this same segfault with two I2C devices:

```
/home/pilothouse/pilothouse/node_modules/segfault-handler/build/Release/segfault-handler.node(+0xd7d)[0xb7741d7d]
[0xb774640c]
/usr/local/bin/node(_ZN2v88internal13GlobalHandles6CreateEPNS0_6ObjectE+0x40)[0x838e9f0]
/usr/local/bin/node(_ZN2v88internal19Runtime_ArrayConcatENS0_9ArgumentsEPNS0_7IsolateE+0x6c2)[0x84be202]
[0x3380a376]
... more ...
```

All that the first one does is `var gyro = new mraa.I2c(1);`. Nothing else was done with the `gyro` variable.  The second I2C bit does 

``` javascript
    var pwm = new mraa.I2c(pwmBus);
    pwm.address(pwmAddress);
    pwm.writeReg(0x00, 0x10); // Set to sleep mode for update to prescale
```

And that's enough to cause the segfault. If I take out the `pwm.writeReg()` line or the `gyro` line it doesn't segfault.

arfoll, 2015-03-12:
What version of mraa are you running? these issues are kind of weird.

arfoll, 2015-03-13:
Sorry but I just don't see the segfault with the following:
```
> > x = > x = nm.I2c(0)> x = nem.I2c(0)> x = newm.I2c(0)> x = new m.I2c(0)        
{}
> x.address> x.address(0xFF)
5
> > x.address(0xFF> x.address(0xF)> x.address(0xF2) 
5
> x.writeReg(0x00, 0x10)
```

Obviously the devices don't exist so it doesn't do much... Can you explain more what you are doing and how you're building mraa?

srlmproductions, 2015-03-14:
Your code is formatted a bit weird, but I think it's supposed to be:

```javascript
var mraa = require('mraa');

x = new mraa.I2c(0);
x.address(0xFF);
x.address(0xF2);
x.writeReg(0x00, 0x10);
```
That doesn't produce an error for me.

I'm running an Edison using mraa v0.6.1-14-g0dd0395 from npm under self-compiled Node v0.10.36 on Ubilinux Debian 7 (wheezy).

I don't really have a concrete idea on why/how this happens, only the program in the first post. It throws the error at line 1152 for me.

arfoll, 2015-05-19:
Try compiling with debug and run with gdb to get a good stacktrace on the latest git master or release version (0.6.2). Instructions are here: https://github.com/intel-iot-devkit/mraa/blob/master/docs/building.md. Since I can't reproduce this is really the only way I can help you.

alext-mkrs, 2015-07-26:
@arfoll, and one more abandoned one. BTW let me know fi you don't want me to do such triaging, IMHO that helps to limit the number of opened issues/requests, which in turn helps to focus on relevant and active issues.

srlmproductions, 2015-07-26:
I've now upgraded to Node v0.12.7 and mraa version 0.7.3 on the Edison. At this point I can't even get past the button line: it keeps saying `Error: Illegal arguments for construction of _exports_Gpio`. I've tried numbers 6, 14, 31, 36, and 48 and none of them work. Which is odd, since those are the numbers from the ISR and GPIO examples.

arfoll, 2015-07-27:
Stop using ubilinux, I can't reproduce on Yocto. This works.

closed 2015-07-27 by arfoll