eclipse-mraa/mraa #162: Node: Kernel Panic in Ubilinux on non-existent I2C address

srlmproductions, 2015-03-11:
I managed to induce a kernel panic from Node. The program:

``` javascript
'use strict';

var mraa = require('mraa'); //require mraa
console.log('MRAA Version: ' + mraa.getVersion()); //write the mraa version to the console

var wind = new mraa.I2c(1);
wind.address(3);

var counter = 0;
setInterval(function () {
    wind.read(2);
    console.log(counter ++);
}, 100);
```
The key point is that there is nothing with the address of 3 on I2C bus 1.

The panic is:
```
Message from syslogd@ubilinux at Mar 11 02:35:02 ...
 kernel:[ 1445.368699] CPU: 1 PID: 2178 Comm: node Tainted: G        W  O 3.10.17-poky-edison-ww42+ #4

Message from syslogd@ubilinux at Mar 11 02:35:02 ...
 kernel:[ 1445.368704] Hardware name: Intel Corporation Merrifield/BODEGA BAY, BIOS 466 2014.06.23:19.20.05

Message from syslogd@ubilinux at Mar 11 02:35:02 ...
 kernel:[ 1445.368711] task: f675de90 ti: f5e8a000 task.ti: f5e8a000

Message from syslogd@ubilinux at Mar 11 02:35:02 ...
 kernel:[ 1445.368772] Stack:

Message from syslogd@ubilinux at Mar 11 02:35:02 ...
 kernel:[ 1445.368841] Call Trace:
0

Message from syslogd@ubilinux at Mar 11 02:35:02 ...
 kernel:[ 1445.369162] Code: b3 ff ff 89 f8 09 d0 80 ce 04 83 ff 02 0f 44 c2 8b 15 54 6b b1 c1 89 82 00 b3 ff ff f7 c6 00 02 00 00 74 14 e8 67 f7 0a 00 56 9d <83> c4 04 5b 5e 5f 5d c3 90 8d 74 26 00 56 9d e8 51 f3 0a 00 83

Message from syslogd@ubilinux at Mar 11 02:35:02 ...
 kernel:[ 1445.369188] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W  O 3.10.17-poky-edison-ww42+ #4

Message from syslogd@ubilinux at Mar 11 02:35:02 ...
 kernel:[ 1445.369193] Hardware name: Intel Corporation Merrifield/BODEGA BAY, BIOS 466 2014.06.23:19.20.05

Message from syslogd@ubilinux at Mar 11 02:35:02 ...
 kernel:[ 1445.369201] task: c1b0e120 ti: c1b08000 task.ti: c1b08000

Message from syslogd@ubilinux at Mar 11 02:35:02 ...
 kernel:[ 1445.369263] Stack:

Message from syslogd@ubilinux at Mar 11 02:35:02 ...
 kernel:[ 1445.369331] Call Trace:

Message from syslogd@ubilinux at Mar 11 02:35:02 ...
 kernel:[ 1445.369575] Code: 8b 42 08 a8 08 75 24 31 c9 8d 42 08 89 ca 0f 01 c8 0f ae f0 89 f6 89 e0 25 00 e0 ff ff 8b 40 08 a8 08 75 07 b1 01 89 f0 0f 01 c9 <85> 1d 58 bd b4 c1 75 0d 8d 55 f0 b8 05 00 00 00 e8 8d 34 d9 ff
```

arfoll, 2015-03-12:
That's abit embarrasing, what breakout board are you using? I can't reproduce here with the arduino breakout using 0.6.1 tag.

srlmproductions, 2015-03-14:
Hmmm. Now I can't reproduce it either. I did switch to homebuilt node, instead of the Debian package. If I get a second Edison I'll try it with the package node, but I'll go ahead and close this for now.

Current environment: Edison using mraa v0.6.1-14-g0dd0395 under homebuilt Node v0.10.36 on Ubilinux Debian 7 (wheezy). Sparkfun Edison blocks: Edison->Console->SD->I2C->GPIO->PWM->LSM9DS0.

closed 2015-03-14 by srlmproductions