eclipse-mraa/mraa #164: Question: Cross compiling, Node build process

srlmproductions, 2015-03-12:
I've been running into some stability issues using mraa for Node (eg #160, #161, #162), and I'd like try and fix them in a fork. But I'm unsure of how to compile MRAA into a `node_module` ready set of files.

So, my questions are:
1. How do I cross compile mraa on my x86_64 Linux desktop for the Edison?
2. How do I produce the files that go into the NPM `node_modules` package, and can I cross compile those too? I've seen [the npm doc](https://github.com/intel-iot-devkit/mraa/blob/master/docs/npm.md), but it wasn't much help.
3. What are the relevant files that I should look at first when debugging Node issues?

arfoll, 2015-03-12:
We'd love to get your fixes :)

Please read https://github.com/intel-iot-devkit/mraa/blob/master/docs/building.md (the bottom includes how to cross compile) & https://github.com/intel-iot-devkit/mraa/blob/master/docs/contributing.md.

1. See building.md that explains how to use the oe toolchain. Lazy people (like me) compile on their board by installing cmake and swig 3.x on their board
2. the npm module won't help you for dev because that uses a pre-generated swig wrapper file. For dev, use cmake.
3. Typically node issues exist in the underlying C code, so usually I build with debug and then use gdb. If there really is an issue in SWIG again gdb through the wrapper file (though that's not for the faint of heart) and the src/mraa.i src/javascript/*.i files. Personally if you're getting kernel panics then it's the kernel's fault - not mraa.

arfoll, 2015-03-12:
Closing - feel free to reopen with Qs about compiling if any of the docs are not clear

closed 2015-03-12 by arfoll