eclipse-mraa/mraa #165: Node: Feature Request: Callbacks

srlmproductions, 2015-03-12:
The standard Node.js paradigm requires that I/O functions take callbacks so that they don't block the main thread. mraa doesn't currently do this. It appears to be primarily a [problem](http://www.swig.org/Doc3.0/Javascript.html) with SWIG itself:

> 26.2.3 Known Issues
>
> At the moment, the Javascript generators pass all tests syntactically, i.e., the generated source code compiles. However, there are still remaining runtime issues.
> - ...
> - Javascript callbacks are not supported

Node addons do, of course, [support callbacks](https://nodejs.org/api/addons.html#addons_callbacks) but it looks like the wrapper code would all have to be written manually. Which might lead to better results with respect to stability, performance, and documentation.

It looks like the recent ISR [commit](https://github.com/intel-iot-devkit/mraa/commit/008184c305cb58f88b2a5fac53ef83eb09e75f5a) takes this approach of writing specialized C++ wrapper code instead of using SWIG. At least I think so, but I'm still learning.

Thoughts?

arfoll, 2015-03-12:
Yes, that is the node 'philosophy'. However most IO calls take such a small amount of time you'd spend more time scheduling work for your thread pool than actually doing work. Specially for the gpio stuff (i2c/spi stuff could be debtated but actually they're async from the kernel pov so again you're just sending tasks). Isr is obviously the exception so we accept some weird code to be able to handle it (in python similar story).

Now realistically I don't see the issue, you can just make your functions run asynchronously, the fact your underlying mraa IO isn't shouldn't really be a concern.

I realise some devs think everything non async will slow them down but in this case it's the other way around :)

closed 2015-03-12 by arfoll

srlmproductions, 2015-03-13:
Can I take you up on that debatable? Take reading 6 bytes from an I2C sensor:

[image]
(image taken from #168; I'm just using it to demonstrate the timing)

That read takes 220uS, from start to finish. Reading 3 bytes takes 146uS (measured). For an IMU type setup, there's three axes to read at least at 100Hz. Taken together, that's 66ms every second waiting for the I/O to finish.

I suppose the real question is could Node make effective use of those 220uS? 

To test that, we have a program:

```javascript
device.writeByte(0x28 | 0x80);
device.read(2);
```
Which produces:
[image]

With this, we can measure the time from the end of the first command to the beginning of the second command. I'm going to assume that this is the cost of returning to Javascript and then back to C++.

The context switching time is 143uS.

My analysis is that callbacks would have performance penalties for reading 1-3 bytes. Above 4 bytes callbacks might be helpful. Maybe. What do you think?

arfoll, 2015-03-13:
Of course - you're more than welcome :) To give you a bit of context the decision was done at the time when we only supported quark platforms which are considerably slower than the edison. But anyways I figured I'd never actually tried it so instead of spewing rubbish at you I should just test it - *for science*!

I made a simplisitc benchmark using rdtsc, a 2 byte read in mraa vs creating a libuv thread (and doing 'nothing') in C (eliminating the noise of v8 etc...), the result is pretty clear, picking an average run you can clearly see using a libuv thread costs you alot more than mraa_i2c_read. Note that uv_queue_work would actually be a bit slower in real life (especially if we actually did 'something' in it :))

```
root@edison:~/mraa/rdtsc-example# ./rdtsc 
0 & 0
91068874609715 & 91068874696760
87045
91068874862885 & 91068875087570
224685
```

Obviously larger reads will eventually bring the difference down etc... and going via the v8 interpreter might bring the cost down/up somehow but at least in C that's the cost. You can check the benchmark here: https://github.com/arfoll/rdtsc-example (you'll have to build your own libuv.so).

If I still haven't convinced you I recommend you have a look at the cylon-js API which is asynchronous (at a js level) and uses mraa. Also I'd love to see a way of making SWIG generate async bindings if someone could point me to somewhere with that I'd be happy to reconsider my current position.

srlmproductions, 2015-03-14:
Interesting. I had no idea that forking and joining a thread was almost 3 milliseconds. That's pretty significant. Thanks for following up.