eclipse-mraa/mraa #166: Help to build addon

rafaneri, 2015-03-12:
Hello everybody,

I need help to build a nodejs addon to create an interface for  Hx711 ADC ([repository](https://github.com/rafaneri/hx711-addon)).

The addon was build, but when I try to run the test script, I get the message: undefined symbol: mraa_gpio_close.

[image]

Thank's for advance!

arfoll, 2015-03-12:
That's awesome, have you looked at submitting this to UPM instead? Also you are aware there is a js interface to mraa (though nothign wrong with a nice native module :))

You don't link against mraa, you need to add -lmraa to the linker flags in binding.gyp something like (if you do ldd against your node module you'll see:
```
brendan@hippo ~/hx711-addon/src (git)-[master] % ldd build/Release/hx711.node 
	linux-vdso.so.1 (0x00007fffc87fe000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007ff260bde000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007ff2608d9000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007ff2606c3000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007ff2604a6000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007ff260103000)
	/usr/lib64/ld-linux-x86-64.so.2 (0x00007ff261133000)
```
but no mraa :(. I've submitted a PR with a fix here:
https://github.com/rafaneri/hx711-addon/pull/1

One comment (had a quick run through the code) https://github.com/rafaneri/hx711-addon/blob/master/src/HX711.cpp#L9 - if you call mraa_init() check the return value otherwise it's a waste of time (it's already getting called on libmraa load)

closed 2015-03-12 by arfoll

rafaneri, 2015-03-12:
@arfoll I'm using the mraa to many sensors, but to this ADC, I did observed that the read time in nodejs was delayed. Then, to do a test, I wrote the native addon. Thanks for the comments, it was my first addon in my life, I going to better it! :D