eclipse-mraa/mraa #167: Incorrect and unexpected return value for getPinCount()

rwaldron, 2015-03-12:
mraa@0.6.1-14-g0dd0395

```sh
root@eddie:~# node
> var mraa = require("mraa");
undefined
> mraa.getPinCount()
56
> mraa.getPlatformName()
'Intel Edison'
```

Which can't be possible, since the Edison is currently attached to the Arduino board, not the Mini board.

rwaldron, 2015-03-12:
A few more notes: 

- The issue occurs when the board is powered by 4 cell (14.8V), 2.0A LiPo that's regulated via a 12V BEC. 
- The issue does not occur when the board is powered by the provided 12V 1.5A DC wall wart
- The issue does not occur when the board is powered by a 2 cell (7.4V), 1A LiPo with no regulator. 

Is it possible that the .5A created a condition that produced the erroneous result?

tingleby, 2015-03-12:
Yes, It is very likely that under those power conditions the peripherals on the arduino board are not being correctly powered.
You can check for gpio 214 when running on that power source, if it doesn't exist then the i2c expanders on the board or not being used. Which are regretably required

arfoll, 2015-03-13:
I don't know enough about the low level hardware for my comments to be worth much but this looks like a power issue - can you check with i2c-detect to see if you can see any of the gpio expanders on your breakout board?

rwaldron, 2015-03-15:
I've now also encountered this issue with

- 12V 1A wall wart
- 12V 1.5A wall wart (the same used in the tests reported above)

```
root@eddie:~# node -e 'console.log(require("mraa").getPinCount());'
56
root@eddie:~# cd /sys/class/gpio/
root@eddie:/sys/class/gpio# ls
drwxr-xr-x    2 root     root             0 Jan  1  2000 .
drwxr-xr-x   52 root     root             0 Jan  1  2000 ..
--w-------    1 root     root          4096 Mar 15 22:31 export
lrwxrwxrwx    1 root     root             0 Jan  1  2000 gpio124 -> ../../devices/pci0000:00/0000:00:0c.0/gpio/gpio124
lrwxrwxrwx    1 root     root             0 Jan  1  2000 gpio125 -> ../../devices/pci0000:00/0000:00:0c.0/gpio/gpio125
lrwxrwxrwx    1 root     root             0 Jan  1  2000 gpio126 -> ../../devices/pci0000:00/0000:00:0c.0/gpio/gpio126
lrwxrwxrwx    1 root     root             0 Jan  1  2000 gpio127 -> ../../devices/pci0000:00/0000:00:0c.0/gpio/gpio127
lrwxrwxrwx    1 root     root             0 Jan  1  2000 gpio128 -> ../../devices/pci0000:00/0000:00:0c.0/gpio/gpio128
lrwxrwxrwx    1 root     root             0 Jan  1  2000 gpio129 -> ../../devices/pci0000:00/0000:00:0c.0/gpio/gpio129
lrwxrwxrwx    1 root     root             0 Jan  1  2000 gpio130 -> ../../devices/pci0000:00/0000:00:0c.0/gpio/gpio130
lrwxrwxrwx    1 root     root             0 Jan  1  2000 gpio131 -> ../../devices/pci0000:00/0000:00:0c.0/gpio/gpio131
lrwxrwxrwx    1 root     root             0 Jan  1  2000 gpio132 -> ../../devices/pci0000:00/0000:00:0c.0/gpio/gpio132
lrwxrwxrwx    1 root     root             0 Jan  1  2000 gpio133 -> ../../devices/pci0000:00/0000:00:0c.0/gpio/gpio133
lrwxrwxrwx    1 root     root             0 Jan  1  2000 gpio134 -> ../../devices/pci0000:00/0000:00:0c.0/gpio/gpio134
lrwxrwxrwx    1 root     root             0 Jan  1  2000 gpiochip0 -> ../../devices/pci0000:00/0000:00:0c.0/gpio/gpiochip0
lrwxrwxrwx    1 root     root             0 Jan  1  2000 gpiochip216 -> ../../devices/pci0000:00/0000:00:08.0/i2c-1/1-0021/gpio/gpiochip216
lrwxrwxrwx    1 root     root             0 Jan  1  2000 gpiochip232 -> ../../devices/pci0000:00/0000:00:08.0/i2c-1/1-0022/gpio/gpiochip232
lrwxrwxrwx    1 root     root             0 Jan  1  2000 gpiochip248 -> ../../devices/pci0000:00/0000:00:08.0/i2c-1/1-0023/gpio/gpiochip248
--w-------    1 root     root          4096 Jan  1  2000 unexport
```

tingleby, 2015-03-16:
Output from `cat /sys/kernel/debug/gpio` would be much more informative.
Do you have any peripherals hooked up during this?
I very much doubt there is anything we can do from a Software standpoint.

arfoll, 2015-03-19:
Closing as this is a hardware issue. To get more exposure I'd recommend the intel support forums.

closed 2015-03-19 by arfoll

rwaldron, 2015-03-19:
I'll keep investigating when I have more time.

rwaldron, 2015-03-27:
I hadn't encountered this issue in a few weeks and then all of a sudden it was there again today. Here's the output requested: 

```
root@rover:~# cat /sys/kernel/debug/gpio
GPIOs 0-191, pci/0000:00:0c.0, 0000:00:0c.0:
 gpio-61  (SW1UI4              ) in  hi
 gpio-64  (bcm43xx_irq         ) in  lo
 gpio-71  (bcm_bt_lpm          ) out lo
 gpio-77  (sd_cd               ) in  hi
 gpio-96  (vwlan               ) out hi
 gpio-111 (Arduino Shield SS   ) out hi
 gpio-124 (hsu                 ) in  hi
 gpio-125 (hsu                 ) in  hi
 gpio-126 (hsu                 ) in  hi
 gpio-127 (hsu                 ) in  hi
 gpio-128 (hsu                 ) in  hi
 gpio-129 (hsu                 ) in  hi
 gpio-130 (hsu                 ) in  hi
 gpio-131 (hsu                 ) in  hi
 gpio-132 (hsu                 ) in  lo
 gpio-133 (hsu                 ) out lo
 gpio-134 (hsu                 ) in  lo
 gpio-184 (bcm_bt_lpm          ) out lo
 gpio-185 (bcm_bt_lpm          ) in  lo

GPIOs 216-231, i2c/1-0021, pcal9555a, can sleep:

GPIOs 232-247, i2c/1-0022, pcal9555a, can sleep:

GPIOs 248-263, i2c/1-0023, pcal9555a, can sleep:
root@rover:~# node -e 'console.log(require("mraa").getPinCount());'
56
```