eclipse-mraa/mraa #169: GPIO - Galileo Gen 2 / I2C

punitganshani, 2015-03-13:
I'm trying to hook up Grove RGB LCD to Galileo Gen 2 using javascript.  Seems like we have to set some GPIO pins 60 and 19 to OUT direction as per the Python script 

https://communities.intel.com/servlet/JiveServlet/download/272893-90734/seti2cGen2.py.zip

I've written few lines of code

[image]

But when I try going it using javascript, I am getting following error

[image]
                                                                                                                  
So if the python script works, shouldn't the Javascript also work?

srlmproductions, 2015-03-13:
IIRC, that error is that you can't use that pin with the requested function. So, there is no GPIO pin 60 as mraa sees it.

In fact, I don't see a pin 60 on the [Intel Galileo Gen 2 pin mappings](http://download.intel.com/support/galileo/sb/galileoiomappingrev2.pdf). I see a pin 19 (the second half of your script). Are you sure both pins are actually outputting?

In addition, mraa does some mapping to standardize pin numbers across different platforms. You'll need to take a look at the [source](https://github.com/intel-iot-devkit/mraa/blob/e8bc227a92fa702bceef5052961123c2e511e811/src/x86/intel_galileo_rev_g.c) for the correct pin number (indicated by the index in `pins[]`).

I think I saw somewhere that you could write something like `new mraa.Gpio(mraa.pins.GPIO14)`, but I can't find anything like that. It would be handy, though.

ps: Instead of taking screenshots of your code and error, you should use [fenced code blocks](https://guides.github.com/features/mastering-markdown/#GitHub-flavored-markdown) (triple back tick) to put the raw text inline.

arfoll, 2015-03-13:
@srlmproductions is 100% correct there is no gpio 60. If you want to use raw gpios pass the raw=True flag in the gpio constructor. Also ignore that script, mraa replaces all those hacks :)

But in this case this isn't what you want to do. You're trying to use i2c to control the grove rgb lcd. Open an i2c handle (I2c(0)) to your i2c bus and use it. Mraa will do the work for you in the background to enable the muxes correctly etc... you don't need to touch that stuff yourself. Just
```
var m = require('mraa');
var b = new m.I2c(0);
// start using my i2c bus
```
FYI there is actually a small js example for the rgb lcd here - https://github.com/intel-iot-devkit/mraa/blob/master/examples/javascript/rgblcd.js

But why are you trying to this by hand anyways? You can use the UPM module to do this for you
```
var LCD = require('jsupm_i2clcd');
var myLcd = new LCD.Jhd1313m1 (0);
```

punitganshani, 2015-03-13:
For raw m.I2c(0) I'm getting this error

[image]

I have tried this options before raising this issue - it did not not work on Galileo Gen 2 board

```
var LCD  = require('jsupm_i2clcd');  
// Initialize Jhd1313m1 at 0x62 (RGB_ADDRESS) and 0x3E (LCD_ADDRESS) 
  
var myLCD = new LCD.Jhd1313m1(0, 0x3E, 0x62);    
console.log('LCD init!');  
myLCD.setCursor(0,0);  
myLCD.write("Hello world!");  
myLCD.setCursor(1,0);  
myLCD.write("Hello world!");  
myLCD.close();  
  
console.log('LCD done!');  

```

Nothing gets printed on LCD, only the color changes if I use setColor.

arfoll, 2015-03-13:
You need your board to be on 5V for the LCD screen to work - try flicking the switch on your grove shield.

Weird mraa direct does that - what version of mraa is it?

punitganshani, 2015-03-13:
mmra => Version 0.6.1

The board is powered with default Galileo Gen 2 power supply..  The code is able to change backlight, but does not show any text.

I also did i2cdetect on Galileo Gen 2, I'm getting this output

```
i2cdetect -l
i2c-0   i2c             intel_qrk_gip_i2c                       I2C adapter
 
i2cdetect -y -r 0
      0  1   2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- UU UU UU -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- --
50: -- -- -- -- UU UU UU UU -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

```

Anything needs to be changed, let me know

arfoll, 2015-03-13:
You're using he grove shield right? That has a 3.3v->5v physical switch on
it. What position is it in?

punitganshani, 2015-03-13:
Ah!  For some reason, it was towards the left.  Flipping it to right, worked.

Cheers!

closed 2015-03-13 by punitganshani