eclipse-mraa/mraa #170: SPI on MBMax with python strangeness

tripzero, 2015-03-14:
Using the latest(as of this writing) mraa master on the minnowboard max I see some strange issues.  The following works:

$python
>> import mraa
>> spi = mraa.Spi(0)

however, if I create a script, it fails (test.py):

#!/usr/bin/env python
import mraa
spi = mraa.Spi(0)

$python test.py
Traceback (most recent call last):
  [snip]
    self.spiDev = mraa.Spi(0)
  File "/usr/lib/python2.7/dist-packages/mraa.py", line 806, in __init__
    this = _mraa.new_Spi(*args)
ValueError: Error initialising SPI bus

However, if I run it with ipython, it works:

$ipython test.py

success!

Syslog documents the unsuccessful and successful attempts:

Mar 14 13:43:48 kev-MinnowBoard-MAX-B3-PLATFORM libmraa[9353]: gpio: Failed to open export for writing
Mar 14 13:43:48 kev-MinnowBoard-MAX-B3-PLATFORM libmraa[9353]: spi: failed to set-up spi sclk multiplexer
Mar 14 13:43:53 kev-MinnowBoard-MAX-B3-PLATFORM libmraa[9354]: libmraa version v0.6.1-5-g3b3b940 initialised by user 'kev' with EUID 1000

This sort of issue screams race condition, but I will defer to the expertise of the mraa team.

arfoll, 2015-03-14:
You need to update mraa that looks like 0.6.1-5 which is not the latest, off the top of my head its -14 you *need* https://github.com/intel-iot-devkit/mraa/commit/c621fad5f3ebbdc0c9398e4f808ed7e379c9af5e. You dont show the full log but my guess is that you didn't install so Ipython grabs the correct module from your current dir but your script does not since its run from another dir.

Does spidev0.0 exist? And does kev have write access to it?

tripzero, 2015-03-14:
> off the top of my head its -14 you *need* c621fad

Odd...  I have installed the latest, which during build time cmake says
is:

INFO - libmraa Version v0.6.1-16-gc621fad

However, syslog still says it's 0.6.1-5.  I'll dig into this more...

arfoll, 2015-03-14:
thought that may be the case - it could be cmake caching version.c sometimes it's own cleverness gets in it's own way :)

Are you sure you're loading the same module from the same place with ipython & your script? Maybe if you call getVersion() from both scripts at the start to check.

tripzero, 2015-03-14:
I was about to try getVersion() :P... After applying PR #171 and running
make uninstall, it appears that I have a rouge version of mraa hanging
around.

tripzero, 2015-03-14:
The reason is on ubuntu, due to a missing patch PR #141 , mraa-python isn't being istalled in a place ubuntu's python can find it.  So a while ago, I had manually copied them...

After patching this mraa with PR #141 and reinstalling, the problem is fixed.

closed 2015-03-15 by arfoll