eclipse-mraa/mraa #172: Problems generating wrappers for Python 3.4.1

ovicin, 2015-03-18:
I try to use mraa with Python 3.4.1 but I have problems in generating the mraa wrappers for this version because the cmake scripts is not detecting it. I tried the following on Intel Edison and Raspberry PI:
1. I compiled Python 3.4.1 from source
```
wget https://www.python.org/ftp/python/3.4.1/Python-3.4.1.tar.xz
tar -Jxf Python-3.4.1.tar.xz
cd Python-3.4.1
./configure --prefix=/opt/python3.4.1 --enable-shared
make
make install
```
2. I simlynked the content from /opt/python3.4.1/lib to usr/lib/
4. I compiled mraa
```
git clone https://github.com/intel-iot-devkit/mraa
cd mraa
mkdir build
cd build
cmake ..
make
```
When I run cmake it detected only Python 2.7 and at the en it generated the python wrappers only for python2.7 (on the Raspberry Pi it also generated the wrappers for python3.2 that was installed from the Raspbian repository)

In the mraa building documentation it says that python3.4 is supported.
Maybe I am missing something, any hints? 

Thanks

arfoll, 2015-03-18:
Try the `tocompile` branch. That has the new python detection (from #141) you should be able to pass `cmake -DBUILDPYTHON3=ON ..` and get the right thing (if that doesn't work please shout since that will get merged in master v.soon). Otherwise using master you'll need to set the following python vars by hand to force the correct python to get found - something like:
```
PYTHON_INCLUDE_PATH /opt/python3.4.1/include/python3.4
PYTHON_LIBRARIES /opt/python3.4.1/lib/libpython3.4.so
PYTHON_EXECUTABLE /opt/python3.4.1/bin/python3.4
PYTHON_INCLUDE_DIR  /opt/python3.4.1/include/python3.4
```

Note that simlinking the interpreter is not enough, that won't work. If using multiple python versions it's worth looking at something like virtualenv which will set all the correct paths/vars for you at a shell level - https://wiki.archlinux.org/index.php/Python_VirtualEnv

arfoll, 2015-03-19:
Closing - please reopen if this is still an isue.

closed 2015-03-19 by arfoll

ovicin, 2015-03-20:
In the end I didn´t managed to compile for Python3.
First I tried using virtualenv, but I don´t know why virtualenv was not copying the shared variables in ```myenv/lib``` so when running ```cmake -DBUILDPYTHON3=ON ..``` is not finding the Python3 libraries.
After that I also tried to rebuild python and install it in the /usr folder
```
./configure --prefix=/usr --enable-shared
make
make install
```
So now in /usr/lib I have the libraries:
```
/usr/lib/libpython3.4m.a
/usr/lib/libpython3.4m.so
/usr/lib/libpython3.4m.so.1.0
/usr/lib/libpython3.so
```
But when running ```cmake -DBUILDPYTHON3=ON ..``` cmake throws the following error:
```
CMake Error at /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:108 (message):
  Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS)
  (Required is at least version "3")
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:315 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-2.8/Modules/FindPythonLibs.cmake:186 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  src/python/CMakeLists.txt:7 (find_package)
```
I used the master branch because I saw that you already integrated there the -DBUILDPYTHON3 flag implementation

arfoll, 2015-03-20:
When you get the error above wipe your build/ directory cmake cache is likely getting upset because FindPython already 'cached' finding python2.

Failing that what distribution are you using?

ovicin, 2015-03-20:
I use the Yocto Intel Edison distribution downloaded about a month ago from the Intel website.

I deleted the content from the build folder before running cmake

arfoll, 2015-03-20:
I got that to work with:
```
cmake -DBUILDPYTHON3=ON -DPYTHON_INCLUDE_DIR=/usr/include/python3.4m/ -DPYTHON_LIBRARY=/usr/lib/libpython3.so ..
```

Then note make install will go to wrong location unless you update the PYTHON_EXECUTABLE var. If that doesn't work dpaste `cmake -LA`

ovicin, 2015-03-20:
Great, the build now works, but unfortunately ```make install``` doesn't install it to the correct path.
See bellow the modules that I have in python3 after running ```make install``` and the output of ```cmake -LA ..```
```
root@vision:~/mraa/build# python3
Python 3.4.1 (default, Mar 19 2015, 23:39:08)
[GCC 4.8.2] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> help ("modules")

Please wait a moment while I gather a list of all available modules...

CDROM               _weakrefset         heapq               runpy
DLFCN               abc                 hmac                sched
IN                  aifc                html                select
TYPES               antigravity         http                selectors
__future__          argparse            idlelib             setuptools
_ast                array               imaplib             shelve
_bisect             ast                 imghdr              shlex
_bootlocale         asynchat            imp                 shutil
_bz2                asyncio             importlib           signal
_codecs             asyncore            inspect             site
_codecs_cn          atexit              io                  smtpd
_codecs_hk          audioop             ipaddress           smtplib
_codecs_iso2022     base64              itertools           sndhdr
_codecs_jp          bdb                 json                socket
_codecs_kr          binascii            keyword             socketserver
_codecs_tw          binhex              lib2to3             spwd
_collections        bisect              linecache           sqlite3
_collections_abc    builtins            locale              sre_compile
_compat_pickle      bz2                 logging             sre_constants
_crypt              cProfile            lzma                sre_parse
_csv                calendar            macpath             ssl
_ctypes             cgi                 macurl2path         stat
_ctypes_test        cgitb               mailbox             statistics
_curses             chunk               mailcap             string
_curses_panel       cmath               marshal             stringprep
_datetime           cmd                 math                struct
_dbm                code                mimetypes           subprocess
_decimal            codecs              mmap                sunau
_dummy_thread       codeop              modulefinder        symbol
_elementtree        collections         multiprocessing     symtable
_functools          colorsys            netrc               sys
_gdbm               compileall          nis                 sysconfig
_hashlib            concurrent          nntplib             syslog
_heapq              configparser        ntpath              tabnanny
_imp                contextlib          nturl2path          tarfile
_io                 copy                numbers             telnetlib
_json               copyreg             opcode              tempfile
_locale             crypt               operator            termios
_lsprof             csv                 optparse            test
_markerlib          ctypes              os                  textwrap
_markupbase         curses              ossaudiodev         this
_md5                datetime            parser              threading
_multibytecodec     dbm                 pathlib             time
_multiprocessing    decimal             pdb                 timeit
_opcode             difflib             pickle              tkinter
_operator           dis                 pickletools         token
_osx_support        distutils           pip                 tokenize
_pickle             doctest             pipes               trace
_posixsubprocess    dummy_threading     pkg_resources       traceback
_pyio               easy_install        pkgutil             tracemalloc
_random             email               platform            tty
_sha1               encodings           plistlib            turtle
_sha256             ensurepip           poplib              turtledemo
_sha512             enum                posix               types
_sitebuiltins       errno               posixpath           unicodedata
_socket             faulthandler        pprint              unittest
_sqlite3            fcntl               profile             urllib
_sre                filecmp             pstats              uu
_ssl                fileinput           pty                 uuid
[... 122 lines cut ...]
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
INFO - libmraa Version v0.6.1-31-g2174ee1
INFO - Target arch is i586
INFO - Adding x86 platforms
-- Found SWIG: /usr/bin/swig (found version "3.0.2")
-- Found PythonLibs: /usr/lib/libpython3.so (found suitable version "3.4.1", minimum required is "3")
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.4.1", minimum required is "3")
WARN - SWIG is 3.0.2. Please upgrade to 3.0.5+ to build nodejs addon
-- Configuring done
-- Generating done
-- Build files have been written to: /home/root/mraa/build
```