eclipse-mraa/mraa #173: Support: DFRobot Romeo and DFRobot IO Expansion Shield

rwaldron, 2015-03-18:
- [Romeo for Edison Controller (With Intel® Edison)](http://www.dfrobot.com/index.php?route=product/product&path=166&product_id=1238)
- [IO Expansion Shield for Intel® Edison](http://www.dfrobot.com/index.php?route=product/product&path=166&product_id=1244)

(Currently neither are in stock)

It seems feasible that they will both work out-of-the-box, being identified as the Arduino Board. If not, I was wondering if support for these could be included in mraa (once they become available, of course).

arfoll, 2015-03-18:
We should definately support those - thanks for pointing these out. In the meantime if someone has one and wants to help please shout!

rwaldron, 2015-03-23:
Update: I added the "Romeo for Edison Controller (Without Intel® Edison)" to the list in the OP.

rwaldron, 2015-06-03:
@arfoll @tingleby 

My order of both of these has just shipped, so presumably they are all back in stock (unless pre-orders wipes them out?)

- http://www.dfrobot.com/index.php?route=product/product&product_id=1241&search=edison&description=true#.VW78SVxViko
- http://www.dfrobot.com/index.php?route=product/product&product_id=1244&search=edison&description=true#.VW782lxViko

arfoll, 2015-06-03:
I still haven't got one of these, hopefully soon. I did however receive one of the xadow edison baords - http://www.seeedstudio.com/wiki/Xadow_-_Edison

rwaldron, 2015-06-03:
@arfoll that's totally awesome, thanks for the link!

rwaldron, 2015-06-13:
Quick update: the [DFRobot Romeo for Intel Edison](http://www.dfrobot.com/index.php?route=product/product&product_id=1241&search=edison&description=true#.VXx-cRNVikp) is being detected as the 56 pin mini breakout.

alext-mkrs, 2015-10-26:
@rwaldron so how did this go, have you been able to run both shields or have you got any problems? Do you think there's a work to be done in mraa for those?

rwaldron, 2015-10-26:
> so how did this go, 

Not very well. I ran out of resource time to work on learning the best way to detect that the Edison is attached to one of these expansion shields. It's necessary to for mraa to know that information because AIO is provided by an I2C IO expander chip on the Romeo. Just testing for some device on the i2c bus with that address is potentially ambiguous. Here's the code they link to from the dfrobot wiki: https://github.com/ouki-wang/remeo4edison

> Do you think there's a work to be done in mraa for those?

Ideally, figuring out how mraa can "know" it's running on either of these, the same way it "knows" that if it's running on Arduino Breakout or Mini Breakout. Currently, it behaves as though it's always Mini Breakout, which won't make sense for the end developer.

jiangzidong, 2016-01-11:
@rwaldron and @arfoll I met the same problem that Romeo is recognized as 'breakout board'. while is can work well with Arduino IDE. Have this problem been solved? or is there some workaround?

codebikeclimb, 2016-01-29:
My Romeo for Edison is no longer programmable from the Arduino IDE. I set it up with Visual Studio and I can cross-compile C++ scripts and run them on the Edison or I can transfer the script via SFTP using WinSCP and compile and run them locally from the Edison's terminal. I am working on trying to use the GPIO & Analog pins from the Arduino breakout, but I can't figure out the mraa pin mapping. Lets say I want to hook up a sensor to digital pins 10 & 11 on the breakout, how do I control those pins through C++ and mraa?

tcassell94, 2017-04-10:
Has there been any work done in this area?

Looking to use the mraa library to work with the Gpio on Edison using the DFRobot IO Expansion [board](https://www.dfrobot.com/product-1244.html) Is it currently possible to do that? I had code working on the Arduino breakout board that does not work at all with the IO Expansion board. Is there something I need to change, or is it not possible at all with the current state of the mraa library?

anzocles, 2017-04-13:
Dear tcassell94,
you can access to any pin of the romeo boar for edison via mraa? I tried as described in
https://medium.com/@monkeytypewritr/control-your-intel-edison-in-realtime-with-websockets-f16e49b2b2c8
... no success. I tried the examples (f.e. the blink programmes) in python, node. Uploading sketches with the arduino ide is working, also running the elf files via exec. (Sorry for that bad English.)

alext-mkrs, 2017-04-13:
The repo link for mraa in that article is way outdated, see the main readme in the repo for an up-to-date one and give it a try.

anzocles, 2017-04-13:
Yes, I've updated to poky 3.5 with the packages in
https://iotdk.intel.com/repos/3.5/intelgalactic-dev/opkg/i586/
the example  mraa/examples/javascript/Blink-IO.js 
does not let the led blink. (Same with the python version  mraa/examples/python/blink-io8.py)
... something I am doing wrong ...

anzocles, 2017-04-14:
may be a permission problem? 
I am using the romeo for edison controller SKU: DFR0331 from dfrobot.
The installed image of poky (more /etc/version): 201606061707
The mraa version (opkg list-installed | grep mraa): mraa - 1.6.1 

When using the "python terminal" (typing python in the shell/terminal) and try to light the LED on pin 13 with a x.write(1) the result is 0 instead of 1 - no light on:
>>> import mraa
>>> x = mraa.Gpio(13)
>>> x.dir(mraa.DIR_OUT)
0
>>> x.write(1)
0

The root cannot do this?
A visudo gives following permission:
root ALL=(ALL) ALL
Hmmm ... seems to be ok. Very strange.

tcassell94, 2017-04-14:
Because these boards aren't supported by mraa (in terms of their pins not being "rerouted" correctly) you must use the raw Edison PIN number. To do this, look at the schematic for the Romeo board (https://www.dfrobot.com/wiki/index.php/Romeo_for_Edison_Controller_SKU:_DFR0331 - first link under development tools) and match the Gpio number to the GP number. For Gpio 13, which is D13 in the schematic, the GP number is GP40. Then, go to the mraa library page for the Edison Mini Breakout board (https://iotdk.intel.com/docs/master/mraa/edison.html) and match the GP number to the mraa number. For GP40 the mraa number is 37. So in your code above, change the pin number to 37 and your code will work on pin 13.

anzocles, 2017-04-15:
Dear tcassell94,
great - it's little bit like treasure hunting with two maps - and it works: the LED is blinking! Hurrah! Thank you!