eclipse-mraa/mraa #175: Lack of the mraa_init call in these example codes

sdssly, 2015-03-19:
Such as the hellomraa.c, helloedison.c, the mraa_init call is needed

michael-ring, 2015-03-19:
Hmm, I guess it it more the opposite, mraa_init is called when the .so is initialized, when you check the return code you will see that it says MRAA_ALREADY_INITIALIZED, so those calls are obsolete.

arfoll, 2015-03-19:
Hi @sdssly, the mraa_init() call is entirely optional (unless your libc is compiled without CTORS). The reason it's not in the examples is because it's really just a safety check and our examples - hopefully obviously - lack alot of error checking in the hope to be clear and to generate decent *interesting* tags in our doxygen docs. Note documentation on mraa_init() hopefully makes this clear - http://iotdk.intel.com/docs/master/mraa/common_8h.html#a1ebdf9bc53d7031e41632b15a17019d3

closed 2015-03-19 by arfoll

michael-ring, 2015-03-19:
Well, documentation seems a little outdated, is says:

Anything but MRAA_SUCCESS should be considered a critical failure

When you call mraa_init in application code you get 

MRAA_ERROR_PLATFORM_ALREADY_INITIALISED

which is not really a critical failure but more a hint that init was already done for you automagically

arfoll, 2015-03-19:
Indeed I re-read that too quickly - fixed documentation with e03eec3ce16fe0a52bc709e259f5b2319739fdd4

sdssly, 2015-03-19:
Thanks for your update! I just found the root cause. For some reason, my local build defined the SWIG. So the mraa_init is not a auto call. 

I would like to know when we will define the SWIG? why use the SWIG?

arfoll, 2015-03-19:
When you do a build the swig modules it uses the %init {} prop in the interface file rather than the constructor https://github.com/intel-iot-devkit/mraa/blob/master/src/mraa.i#L27

How are you building? And why is SWIG defined? It should not be defined when building the C/C++ library only when building the SWIG interface wrapper files.