eclipse-mraa/mraa #176: PWM method period_ms() returns error in python

cosscat, 2015-03-21:
Hello,

While performing some tests on Galileo v1,  for PWM,  period_ms() returns 4 (ERROR_INVALID_PARAMETER = 4).

>>> p.period_ms(2)
0
>>> p.period_ms(3)
0
>>> p.period_ms(4)
0
>>> p.period_ms(9)
4
>>> p.period_ms(8)
4
>>> p.period_ms(7)
0

At the same time, 
>>> p.period_us(900)
0
, so this is not a platform limitation

>>> m.getVersion()
'v0.6.0'

Costin

cosscat, 2015-03-21:
I also observed this behavior for setting the period in us:

>>> p.period_us(10000)
4
>>> p.period_us(1000) 
0
>>> p.period_us(9000)
4
>>> p.period_us(6000)
0
>>> p.period_us(7000)
0
>>> p.period_us(8000)
4
>>> p.period_us(7999)
4
>>> p.period_us(7001)
0
>>> p.period_us(100000)
4
 Same platform Galileo v1

Costin

cosscat, 2015-03-21:
Hello,

After having a look on mraa/src/pwm/pwm.c i saw there is a pwm_min_period and a pwm_max_period that further more are initialized in mraa/src/x86/intel$(machine_name).c. It would me nice to have a function inside Python's mraa Pwm class that can return these two values.

Costin

arfoll, 2015-03-23:
We should provide a way of exporting the min/max pwm period values of a platform

commit 64f962d1 (2015-03-24): pwm: add calls for getting PWM max and min period

Resolves #176
closed 2015-03-24 by tingleby