eclipse-mraa/mraa #179: SPI:  write_word() seems to return wrong values

cosscat, 2015-03-26:
Hello,

This is for Galileo v1 and >>> m.getVersion()   --->  'v0.6.1'

I have a TMP125 SPI temperature sensor (http://www.datasheetcatalog.com/datasheets_pdf/T/M/P/1/TMP125.shtml) and did a small script to test if write_word() works well. To do this i did two scenarios where i read the same register of the sensor twice: once with write_word() and the second time with write(). In the case of write() result seems ok, while for write_word() is not.

```python

#!/usr/bin/env python
import mraa as m
from time import sleep

spi=m.Spi(0)
cs=m.Gpio(0)
cs.dir(0) # set as out
cs.write(1)
sleep(0.2)
#init done

def form_full_byte(no): # used to form full 8 bit bytes
  b=bin(no)[2:] # transforming the int no. to a bin
  if len(b) < 8:
    fb = "0"*(8-len(b)) + str(b) # forming a full byte
    return fb
  else:
    fb = b
    return fb

if __name__ == "__main__" :
  print "Reading from sensor ... \n"
  cs.write(0) # begin to read as cs is set to LOW
  sleep(0.2)
  result = spi.write_word(0)
  result_bin = bin(result)[2:]
  print "write_word result in binary: " + str(rezult_bin)
  raw = bytearray(2)
  var = bytearray(2)
  cs.write(1) # force TMP125 to do a new reading this time with the write func
  sleep(0.2)
  cs.write(0) # begin to read as cs is set to LOW
  sleep(0.2)

  raw = spi.write(var)

  print "Binary res obtained with write: " + form_full_byte(raw[0]) + form_full_byte(raw[1])

```

The idea is that for the temperature range of a room the first bit should be 0 and no matter of the room temperature the last 3 bits should be as  the LSB of the measurement. The measurement is 10 bits (14:5).

Here is the result of two running cycles:

root@galileo:~# ./TMP125.py
Reading from sensor ...

write_word result in binary: 1100000000001100
Binary res obtained with write: 0000110011000000
root@galileo:~# ./TMP125.py
Reading from sensor ...

write_word result in binary: 1100000000001100
Binary res obtained with write: 0000110011000000

Temperature in the room about 24 deg Celsius

Costin

michael-ring, 2015-03-26:
The problem is that spi is per default configured for 8 bit transfers. The write word function should throw an error in this situation as it cannot correctly transfer 16 bits in 8 bit mode, the cs needs to stay low for all 16 bits, this is not possible in 8 bit mode, cs will go up after 8 bits an then down again.
You do your own CS, but this is a special case.

Please call mraa_spi_bit_per_word with 16 as parameter before doing the write_word , either the call fails because Galileo does not support this (I do not have a Galileo, so I do not know) or the call should work after enabling 16bit mode.

cosscat, 2015-03-27:
In the case of CS, how can i do a CS from inside SPI? The SPI connector (compatible with Arduino ones) has 6 pins, and none of them are (from what i understand) related to CS.

arfoll, 2015-03-27:
Use pin 10. Agreed the 6 pin connector is confusing imho best avoided

cosscat, 2015-03-27:
Just did a test, and at this time I am confused ... my sensor is throwing wrong values on the screen that are not compatible with the datasheet itself. I will try to change it. However, I tried with pin 10 and result is the same. Why is there a pin (in this case 10) assigned for CS? If i have 4 devices how can i select each one of them?
Just to continue the test itself, I'll use two Galileos to send/receive data.

arfoll, 2015-03-27:
Because that's the hardware CS. Technically using a GPIO for CS can have unintended consequences - see software CS debate. Techincally on galileo you only have one CS so you can't have 4 devices.

It's possible your sensor can't handle 16bit transfers.

michael-ring, 2015-03-27:
I briefly checked the Datasheet, the Chip seems 16Bit. cosscat, please have a look at the max7219 example, this should give you an idea how to do 16bit transfers with mraa, the max7219 is also 16bit.

@arfoll : Any news on merging the soft-cs pull request?

alext-mkrs, 2015-06-14:
@cosscat have you been able to run your program since then? Do you have any further troubles/questions within this thread's topic, or it can be closed?

alext-mkrs, 2015-07-26:
@arfoll, one more candidate for closure

closed 2015-07-27 by arfoll