eclipse-mraa/mraa #183: The getPinName() routine is absent

dyosick, 2015-04-01:
According to the documentation to the 0.6.1 release (http://iotdk.intel.com/docs/master/mraa/namespacemraa.html#aa436b58bf20a87909c7344bf58d42706), but the compiler says:
"aio.c: In function ‘int main()’:
aio.c:6:38: error: ‘getPinName’ is not a member of ‘mraa’"

tingleby, 2015-04-07:
What version of mraa do you have installed?
`opkg info libmraa0`

arfoll, 2015-04-07:
More importantly this is not a feature of v0.6.1 but is a feature for
upcoming releases:
http://iotdk.intel.com/docs/mraa/v0.6.1/namespacemraa.html#aa436b58bf20a87909c7344bf58d42706

You need to use master from intelgalactic-dev to get that function.

dyosick, 2015-04-08:
Sorry, guys, it's my fault. I've looked into the wrong doc page. Thanks for help.

closed 2015-04-08 by dyosick