eclipse-mraa/mraa #187: Python API's I2c.writeReg() Does Not Produce Expected Results

pnovotnak, 2015-04-08:
When using the Python I2c module, I don't get the results I expect with writeReg().

My `/proc/version` reads

    Linux version 3.10.17-poky-edison+ (sys_dswci@tlsndgbuild004) (gcc version 4.8.2 (GCC) ) #1 SMP PREEMPT Fri Jan 30 14:16:35 CET 2015`

Not only does most of the mraa.I2c module seemingly do nothing, it produces no meaningful error messages either, making it nigh impossible to debug. The writeReg() function returns the integer 5 (?).

    #!/usr/bin/python

    import mraa

    for i in range(0, 8):
        bus = mraa.I2c(i)
        bus.address(0x40)
        bus.writeReg(0xfc, 0xff)

Here's a little program to loop over all of the possible busses and write to the address I want... Perhaps someone here can enlighten me: What am I doing wrong?

FYI, the chip in question is a PCA9685. It's address is 0x40.

The output from `i2cdetect -y -r 1` looks like:

         0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
    00:          -- -- -- -- -- -- -- -- -- -- -- -- --
    10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
    20: UU UU UU UU -- -- -- -- -- -- -- -- -- -- -- --
    30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
    40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
    50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
    60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
    70: 70 -- -- -- -- -- -- --

The output from `i2cdump -y 1 0x40` looks like:

    No size specified (using byte-data access)
         0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
    00: 01 04 e2 e4 e8 e0 00 00 a1 00 00 00 a1 00 00 00    ??????..?...?...
    10: a1 00 00 00 a1 00 00 00 a1 00 00 00 a1 00 00 00    ?...?...?...?...
    20: a1 00 00 00 a1 00 00 00 a1 00 00 00 a1 00 00 00    ?...?...?...?...
    30: a1 00 00 00 a1 00 00 00 a1 00 00 00 a1 00 00 00    ?...?...?...?...
    40: a1 00 00 00 a1 00 XX XX XX XX XX XX XX XX XX XX    ?...?.XXXXXXXXXX
    50: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    60: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    70: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    80: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    90: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    a0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    b0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    c0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    d0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    e0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    f0: XX XX XX XX XX XX XX XX XX XX 00 00 00 00 79 00    XXXXXXXXXX....y.

After I run `i2cset -y 1 0x40 0xfc 0xff` it looks like:

    No size specified (using byte-data access)
         0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
    00: 01 04 e2 e4 e8 e0 00 00 ff 00 00 00 ff 00 00 00    ??????..........
    10: ff 00 00 00 ff 00 00 00 ff 00 00 00 ff 00 00 00    ................
    20: ff 00 00 00 ff 00 00 00 ff 00 00 00 ff 00 00 00    ................
    30: ff 00 00 00 ff 00 00 00 ff 00 00 00 ff 00 00 00    ................
    40: ff 00 00 00 ff 00 XX XX XX XX XX XX XX XX XX XX    ......XXXXXXXXXX
    50: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    60: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    70: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    80: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    90: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    a0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    b0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    c0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    d0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    e0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    f0: XX XX XX XX XX XX XX XX XX XX 00 00 00 00 79 00    XXXXXXXXXX....y.

And I can see the change in an oscilloscope, so I know that those programs work.

pnovotnak, 2015-04-08:
Tried upgrading per instructions in readme which went fine...

    Upgrading libmraa0 on root from 0.5.2-r0 to 0.6.1+git0+805d22f0b1-r0...
    Downloading http://repo.opkg.net/edison/repo/core2-32/libmraa0_0.6.1+git0+805d22f0b1-r0_core2-32.ipk.
    Removing obsolete file /usr/lib/libmraa.so.0.5.2.
    Configuring libmraa0.

Proof-

    >>> mraa.getVersion()
    'v0.6.1'

But still no dice

arfoll, 2015-04-08:
You don't mention what breakout board you're using but I guess you're needing i2c-1, mraa respects the krenel i2c bus numbering so just ask for I2c(1). I suspect you haven't looked at the debugging page http://iotdk.intel.com/docs/master/mraa/debugging.html which explains how to use journalctl to get debug messages from syslog. If you use printError(5) you'll get the error message which is invalid handle - likely because mraa 0.5.2 does not support anything but the arduino breakout. I suspect on 0.6.1 the error is different at least

What does the following give you?
```
import mraa
mraa.getVersion()
x = m.I2c(1)
type(x)
x.address(0x40)
mraa.printError(x.writeReg(0xfc, 0xff))
```

pnovotnak, 2015-04-08:
It gives me `MRAA: Invalid Handle.`

It's a Sparkfun DEV-13042 board

pnovotnak, 2015-04-09:
As for syslog... Looks like that code doesn't output anything there. Alls I see is right after boot:

[   14.223182] wl_bss_connect_done succeeded with 58:6d:8f:56:83:a8
[   14.293219] wl_bss_connect_done succeeded with 58:6d:8f:56:83:a8
[   15.772075] ip (330) used greatest stack depth: 5344 bytes left

arfoll, 2015-04-10:
Sorry for the delay, I've had a quick look - it seems like the kernel i2c driver is refusing to write to 0x40 for some reason. I'm thinking the edison arduino breakout must use the same module or something and the kernel is getting confused. It'll need a further look.

pnovotnak, 2015-04-10:
The breakout board has a few places to bridge the chip to change it's address. I'll give that a shot and post results.

pnovotnak, 2015-04-10:
Disabled the all call address (0x70), moved the chip to 0x40 and rebooted the Edison with the same result as before. I'll take a look at the debugging document and see what information I can collect.

arfoll, 2015-04-10:
Very weird - btw you're not looking at syslog use `journalctl -f` whilst running your calls you should at least get one message from mraa on initialisation.

pnovotnak, 2015-04-10:
D'oh!

Ok, the relevant lines from `journalctl -f` are...

`libmraa[382]: Above i2c bus count` for 0 <= bus number <= 9

Between 0 and 8 I get `libmraa[382]: Invalid i2c bus, moving to default i2c bus` *except for 6, which appears to run successfully.*

After that, no matter what bus number I choose, I get `libmraa[382]: i2c: Failed to write` when writing to an address.

alext-mkrs, 2015-06-04:
@pnovotnak, is this still actual?

Here's a working program (though in C) for this expansion board: https://communities.intel.com/thread/63552

It uses I2C bus #1 and address 0x40 and looks like it worked back in April - so should work with any recent mraa version.

alext-mkrs, 2015-07-26:
@arfoll, I propose to close this one as no response for a while and a reported solution provided.

closed 2015-07-27 by arfoll

fryguy1013, 2016-01-04:
I am seeing this problem as well. No matter what I do, I get `Jan 04 00:26:44 edison libmraa[583]: i2c: Failed to write` in the journalctl log, when doing it in either c++ or python. What's interesting is the i2cdump seems to be able to communicate with the board since it looks like the values I would expect:

    root@edison:~# i2cdump -y 1 0x40
    No size specified (using byte-data access)
         0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
    00: 11 04 e2 e4 e8 e0 00 00 00 10 00 00 00 10 00 00    ??????...?...?..
    10: 00 10 00 00 00 10 00 00 00 10 00 00 00 10 00 00    .?...?...?...?..
    20: 00 10 00 00 00 10 00 00 00 10 00 00 00 10 00 00    .?...?...?...?..
    30: 00 10 00 00 00 10 00 00 00 10 00 00 00 10 00 00    .?...?...?...?..
    40: 00 10 00 00 00 10 XX XX XX XX XX XX XX XX XX XX    .?...?XXXXXXXXXX
    50: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    60: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    70: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    80: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    90: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    a0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    b0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    c0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    d0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    e0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
    f0: XX XX XX XX XX XX XX XX XX XX 00 00 00 00 1e 00    XXXXXXXXXX....?.

I'm using the Sparkfun DEV-13042 (PWM block) attached to the Sparkfun DEV-13097 (Edison Arduino breakout)

arfoll, 2016-01-04:
Please don't use old bugs this bug is more than likely a failure to use bus 1 or 6.

In your case what mraa version and can you share your code? Make sure your arguments to writebyte are the correct way round. Open a new bug with a code snippet if you still have an issue.