eclipse-mraa/mraa #198: I2C bus_id is incorrect on DE3815?

yongli3, 2015-04-29:
I am testing the MRAA on NUC DE3815. I found the custom solutions header is using i2c-8 and i2c-9. It is not 0 and 1:
/hellomraa
hello mraa
 Version: v0.6.2-7-gf6e87f6
 Running on Intel DE3815

i2cdetect -l
i2c-0   i2c             i915 gmbus ssc                          I2C adapter
i2c-1   i2c             i915 gmbus vga                          I2C adapter
i2c-2   i2c             i915 gmbus panel                        I2C adapter
i2c-3   i2c             i915 gmbus dpc                          I2C adapter
i2c-4   i2c             i915 gmbus dpb                          I2C adapter
i2c-5   i2c             i915 gmbus dpd                          I2C adapter
i2c-6   i2c             DPDDC-B                                 I2C adapter
i2c-7   i2c             DPDDC-C                                 I2C adapter
i2c-8   i2c             i2c-designware-pci                      I2C adapter
i2c-9   i2c             i2c-designware-pci                      I2C adapter

i2cdetect -y -r 8
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- 77

tingleby, 2015-04-29:
Would it be possible to get the kernel version please `uname -a` and what distro (if any) you are running.
This is something I was afraid of when I added support for this platform.

yongli3, 2015-04-29:
Thanks for your update. I am using the 3.19 kernel:
Linux DE3815 3.19.0 #27 SMP PREEMPT Thu

tingleby, 2015-04-30:
Ok, When I get chance ill create a patch to try and avoid having I2C statically assigned to 0/1 and move to where it sees the appropriate device

commit cf8817d5 (2015-06-30): x86: de3815: detect correct i2c-dev

Will search for the designware i2c devices
Closes #198
Closes #203
commit 64415fc3 (2015-06-30): x86: de3815: detect correct i2c-dev

Will search for the designware i2c devices
Closes #198
Closes #203
commit 5ccd7a67 (2015-06-30): x86: de3815: detect correct i2c-dev

Will search for the designware i2c devices
Closes #198
Closes #203
closed 2015-06-30 by tingleby