eclipse-mraa/mraa #204: Analog reading error

nourshirazi, 2015-05-09:
Hello,

Since i upgraded the libmraa i am get an error reading analog pins on the Edison breakoutboard.

The error given:
                                                                                                                                                                            
ERROR: /home/root/.node_app_slot/main.js:22                                                                                                                                 
ERROR: var analogPin0 = new mraa.Aio(0); //setup access analog input Analog pin #0 (A                                                                                       
                 ^                                                                                                                                                          
ERROR: Error: Invalid AIO pin specified - do you have an ADC?                                                                                                               
    at Object.<anonymous> (/home/root/.node_app_slot/main.js:22:18)                                                                                                         
    at Module._compile (module.js:456:26)                                                                                                                                   
    at Object.Module._extensions..js (module.js:474:10)                                                                                                                     
    at Module.load (module.js:356:32)                                                                                                                                       
    at Function.Module._load (module.js:312:12)                                                                                                                             
    at Function.Module.runMain (module.js:497:10)                                                                                                                           
    at startup (node.js:119:16)                                                                                                                                             
    at node.js:906:3 

How can this be fixed. is there any way to 'downgrade' mraa.

tingleby, 2015-05-10:
Hi,  You can install previous version of mraa for node with npm
For example 
`npm install mraa@0.6.1`
You can find all the versions of mraa using `npm show mraa`

In this case, are you using anything attached to A4 or A5? or I2C?

tingleby, 2015-06-04:
Hello, Im strugling to reproduce on v0.7.1+ could you try with the latest version.

closed 2015-06-22 by tingleby