eclipse-mraa/mraa #205: Galileo v2 i2C module can't be initialized

cosscat, 2015-05-12:
Hello,

While performing some tests on I2C module, a strange thing happened  when initializing the port 0 (which i am sure is correct since it worked 20 min. ago):
 >>> i2c = m.I2c(0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/mraa.py", line 895, in __init__
    this = _mraa.new_I2c(bus, raw)
ValueError: Invalid i2c bus
>>> 

root@galileo:~# journalctl -f
-- Logs begin at Tue 2015-05-12 15:38:00 UTC. --
May 12 15:39:06 galileo libmraa[189]: libmraa version v0.6.2 initialised by user 'root' with EUID 0
May 12 15:39:27 galileo kernel: intel_qrk_gip 0000:00:15.2: controller timed out
May 12 15:39:27 galileo kernel: pca953x 0-0027: failed writing register
May 12 15:39:27 galileo libmraa[189]: i2c: Failed to set-up i2c sda multiplexer

This started to appear today. Until now, I didn't encounter this.

Doing a search on the web, i found this: https://communities.intel.com/thread/60432  , and if true it means theoretically there is nothing wrong with the board.

Thank you,

Costin

tingleby, 2015-05-16:
So the line 
`May 12 15:39:27 galileo kernel: pca953x 0-0027: failed writing register`
suggest that the multiplexers which are on another I2C bus have crashed,

cosscat, 2015-05-17:
Thank you tingleby. Unfortunatelly, I have an other Galileo v2 that has the same issue too :(
Not  a good sign for me ...

cosscat, 2015-05-20:
Hi tingleby,

Please close the issue. It is something related to the hardware. It appears randomly.

Costin

closed 2015-05-20 by arfoll