eclipse-mraa/mraa #206: pwm->read() failure

dandyd449, 2015-05-16:
I've been unable to read the current duty cycle percentage.
```
#include "mraa.hpp"
#include <unistd.h>

int main(){
    mraa::Pwm* led;

    led = new mraa::Pwm(3);
    if(led == NULL){
        return MRAA_ERROR_UNSPECIFIED;
    }

    led->enable(true);

    led->write(0.9);
    fprintf(stdout, "%f\n", led->read());

    sleep(2);

    led->write(0.01);
    fprintf(stdout, "%f\n", led->read());

    sleep(2);

    led->enable(false);
    led->~Pwm();
}
```

This should: 
Turn on the led
Set the duty cycle to 90%
Set the duty cycle to 1%
Turn off the led

This seems to work but I am unable to read the duty cycle. Logs from my edison show this:
```
libmraa[2288]: [[1;31mpwm: Error in reading period[[0m
```

cosscat, 2015-05-16:
Hello dandyd449,

I had a look on mraa/src/pwm/pwm.c (after cloning mraa) and i saw there that the period is read via:
...
snprintf(bu, MAX_SIZE, "/sys/class/pwm/pwmchip%d/pwm%d/period", dev->chipid, dev->pin);
...
int period_f = open(bu, O_RDWR);
...
ssize_t rb = read(period_f, output, size + 1);

I don't have an Edison at hand, but maybe there is an issue with file permission (i remember having this issue once)

Costin

dandyd449, 2015-05-16:
Hmm... So it is starting to sound more like an issue with the device than with the mraa library... Thats sad for such a capable device.

tingleby, 2015-05-16:
So, I have an rough idea where this could be happening. As the Edison PWM devices all share a period value, a quirk of the hardware, it may have to be read from another file. 
Ill look at this in the next couple of days.

dandyd449, 2015-05-16:
Hmm, not that I've had any reason to change the period yet but you're saying that PWM is true hardware PWM vs say the raspberry pi's software based PWM?

tingleby, 2015-05-16:
So the PWM is hardware based but it isn't terribly flexible. All 4 PWM channels share a period value which is set in a way not following the Linux sysfs interface for PWM.

dandyd449, 2015-05-19:
I can confirm that I can cat the file needed to read the period file.
```
/sys/class/pwm/pwmchip%d/pwm%d/period
```

tingleby, 2015-06-04:
Yeah this doesnt seem to be a problem with me
```
>>> a = mraa.Pwm(3)
>>> a.enable(True)
0
>>> a.write(0.5)
0
>>> a.read()
0.5
>>> a.write(0.01)
0
>>> a.read()
0.0099997995421290398
```
Im going to close the Issue but if you can check against the latest version of mraa and edison image. If it still occurs I will re-examine the issue.

closed 2015-06-04 by tingleby

dandyd449, 2015-06-05:
Please check my other issue that needs reopened. 
It also appears that you are using python. My issue is obviously with the C implementation.

tingleby, 2015-06-05:
The implementation is the same, We use SWIG to hook into the c++ headers like you used.
Ive found the problem #207 is reopened until its fixed.

dandyd449, 2015-06-06:
I upgraded all of my kernel modules using:
```
http://repo.opkg.net/edison/repo/edison/
```
I then ran your python code and found that it worked correctly. I then ran my C code to find that it was still broken.
My next step was to clone mraa master branch and compile with that to ensure it wasn't already fixed. It was not. 

I certainly understand not wanting to try and diagnose an mraa bug knowing there are kernel issues but I'm not convinced it solely a kernel problem. Why would python work and C++ not? 

I'm starting to believe #207 is an issue with the kernel and this issue is with mraa.

alext-mkrs, 2015-06-06:
As @tingleby said that's really strange - the code being executed is the same at the end of the day, Python, C++ and JavaScript are just bindings to the C core.

I've just ran your C++ code posted at the beginning of this thread and it worked fine:
```
root@edison1:~# g++ -lmraa mraa-207-test.cpp -o mraa-207-test
root@edison1:~# ./mraa-207-test 
0.900000
0.010000
```

as well as the Python one posted by @tingleby
```
>>> import mraa
>>> a = mraa.Pwm(3)
>>> a.enable(True)
0
>>> a.write(0.5)
0
>>> a.read()
0.5
>>> a.write(0.01)
0
>>> a.read()
0.0099997995421290398
```

I'm running Release 2.1 image (the latest one, it has "weekly-146" in `/etc/version`) and MRAA 0.6.2 from the intergalactic repo.

@dandyd449, you mentioned that you have updated kernel modules from my repo - have you also installed the kernel from there, or only modules over the official Rel 2.1 kernel? What's your `uname -a` output?

dandyd449, 2015-06-06:
@alext-mkrs 
I had not updated the kernel image. I added the repo and did a "list-upgradable".

I just updated to your kernel and compiled directly on the edision as you did and can confirm that I can now read the pin. Very strange... Thanks for the help! 

@tingleby 
Sorry, for the bad report and thanks for all the great work!

alext-mkrs, 2015-06-06:
@dandyd449 I see, yes, `list-upgradable` would list kernel modules but not the kernel itself (and it would anyway need some tinkering to get updated).

And having kernel modules installed with a different kernel (when they're not compiled together) may lead to such hard to debug and subtle glitches. Generally the kernel in my repo is the same as the official one source-wise, but due to the fact I'm enabling additional modules and features there, the configuration is different, and that might result in such effects sometimes.

dandyd449, 2015-06-06:
I didn't install your modules until this morning so I don't believe that was the problem. I'm starting to believe its either my build environment (as I'm cross compiling using the SDK) or remote GDB isn't working as it should.

alext-mkrs, 2015-06-07:
Understood, thanks for clarification.